PX4/PX4-Autopilot #5261: Camera trigger on-command trigger is broken

mhkabir, 2016-08-06:
Lots of changes for PWM control went in, and something seems to have broken GPIO triggering.

Running
```camera_trigger stop``` and ```camera_trigger start``` via mavlink NSH (yay!) gets it working again, so the functionality is OK, but it doesn't enable on command, even though the shell ```info``` shows it as being "enabled"

LorenzMeier, 2016-08-06:
Thanks for reporting and sorry for the breakage. It would be fantastic if you could have a look. Seems like it should be an easy fix.

mhkabir, 2016-08-06:
Found it. FMU driver clashes with it for some reason if started after trigger. It works if I start the trigger after the FMU driver.

mhkabir, 2016-08-23:
This is fine on master because the trigger pins are set to AUX 5,6 (handled by ```fmu mode_pwm4```). I was having the issue because of my custom config which had them set on 1,2. 

Should be fine as long as people stick to the defaults.

closed 2016-08-23 by mhkabir