PX4/PX4-Autopilot #5264: LPE: Add flow position offset

LorenzMeier, 2016-08-06:
@jgoppert Can you copy the logic from INAV to allow users to enter offset parameters for the flow module so they compensate for rotational motion introducing translations?

jgoppert, 2016-08-07:
I had a look at the compensation. That equation only works assuming a small field of view. If you  have a fov resulting in a distance on ground of 10 cm, then a 10 cm shift from center, will have an affect (rotation will look like translation), but if you have a fov of 1 m, then a 10 cm shift will have virtually no effect and the compensation will hurt you. At high altitude, you will  really be messing up the vehicle with this kind of correction, and I am flying at 5 m with flow now successfully.

I could model the fov effect mathematically (would need flow fov param as well) to fix this if we believe it is necessary for low altitude flight performance, at high altitude, this entire effect is negated. From flight testing, even with flow off center by about 20 cm, I haven't seen a large effect at low altitude and it might add more complication for users.

@ecmnet @mhkabir  Any input?

I will be working on flow tomorrow in flight and sitl, so I'll get some good data on this.

mhkabir, 2016-08-07:
I agree with James on this one, and we could probably model the FOV effect and compensate for it. But I don't think that SITL will be a good candidate for tests here since the actual PX4Flow hardware vs. simulated camera is quite different, esp. FOV wise and algorithm wise.

LorenzMeier, 2016-08-07:
@jgoppert Any robust flow solution has a very small FOV. In particular PX4 Flow has just a couple of degrees.

LorenzMeier, 2016-08-07:
And at 1 m altitude we have seen strong toiled bowling effects without.

jgoppert, 2016-08-07:
@LorenzMeier OK,  I will add it for 1.5.

LorenzMeier, 2017-04-10:
I'm closing this as stale.

closed 2017-04-10 by LorenzMeier