PX4/PX4-Autopilot #5266: SITL unstable on jenkins CI

jgoppert, 2016-08-06:
The sitl test on jenkins has been unreliable since the PR for sitl path fixes. Even after reverting this commit, it still fails on 95% of PR's.

@LorenzMeier @AndreasAntener  @tumbili @julianoes : Any ideas?

dagar, 2016-08-06:
This might be a good excuse to expand the error reporting. How close did it come to finishing successfully? Performance on EC2 can be highly variable.

jgoppert, 2016-08-06:
The best I can tell it is 15 m off of where it should be.  pos_d: 15.013240, alt_d: 0.000000

@dagar I agree. We need to create jpg on jenkins of each flight.

LorenzMeier, 2016-08-06:
Here is how build 633 of master looks after it passed:
http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL/633/
http://logs.uaventure.com/view/qZwMF8ZAwhF4UxSLHhJtYm

Here is how build 634 of master looks after it failed:
http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL/634/
http://logs.uaventure.com/view/gHAjvDpj2FrYMGKenr25Ph

Looks like an estimator failure. Most likely due to timing. We have half-lockstep enabled in SITL (the flight code will keep waiting for a delayed / slow sim). We might need full lockstep (the sim waits for delayed flight code).

Given that this is an estimator failure though (which should not happen in half lockstep) it might be something entirely different.

LorenzMeier, 2016-08-06:
BTW: I don't think Jenkins should generate PNGs - instead it should upload its log file and post the links here as comments.

LorenzMeier, 2016-08-06:
Well, so it turns out that we had used a tier with CPU credits which means that if our usage went up we were starving on CPUs once we were out of credits.

Which sort of means the more PRs went in, the more failures you would see. I've upgraded the instance now to a compute instance which is twice as expensive but should not toy around.

jgoppert, 2016-08-06:
@LorenzMeier awesome, thanks!

closed 2016-08-06 by jgoppert