PX4/PX4-Autopilot #5270: LPSP.Z decrease or increase in the vtol front transition period problem.

CornerOfSkyline, 2016-08-07:
There is a strange  behaviour in the vtol front transition in auto mission. When the previous waypoint's altitude is different with the vtol transilate command point(one normal waypoint + command point) and next waypoint's altitude. The LPSP.Z will increase (previous is higher) or descrease (previous is lower) in the transition period. The LPSP.Z setpoint will cause the mc controller to control the vtol altitude increase or decrease in transition.  The attachment is the sim mission file.

sanderux, 2016-08-08:
Confirmed in SITL, so far ruled out FOH update and i don't see any such behavior commanded from navigator. switching position estimator doesnt seem to help

AndreasAntener, 2016-08-12:
Fixed on master

closed 2016-08-12 by AndreasAntener