PX4/PX4-Autopilot #5276: flashing master wiped parameters on fmu-v2

kd0aij, 2016-08-08:
console log 
```
 sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #2
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
WARN  [px4io] check CRC failed: -22
INFO  [px4io] default PWM output device
[PX4IO] using firmware from /etc/extras/px4io-v2.bin
[PX4IO] found bootloader revision: 4
[PX4IO] erase...
[PX4IO] programming 45988 bytes...
[PX4IO] verify...
[PX4IO] update complete
WARN  [px4io] CRCs match
INFO  [dataman] Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V2
WARN  [hmc5883] no device on bus 2
read_reg failWARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 1
MPU6000 on SPI bus 1 at 4 (1000 KHz)
WARN  [mpu6000] no device on bus 4
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf1xx: command not found
WARN  [commander] No RGB LED found at /dev/rgbled0
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V2
px4flow [183:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Multicopter 
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 
WARN  [local_position_estimator] starting

NuttShell (NSH)
nsh> INFO  [local_position_estimator] [lpe] baro init 1627 m std 148 cm
INFO  [local_position_estimator] [lpe] z resume 
WARN  [mavlink] Disabling hardware flow control
WARN  [mavlink] Disabling hardware flow control
WARN  [commander] Not ready to fly: Sensors not set up correctly
ps
PID   PRI SCHD TYPE   NP STATE    NAME
    0   0 FIFO TASK      READY    Idle Task()
    1 192 FIFO KTHREAD   WAITSIG  hpwork()
    2  50 FIFO KTHREAD   WAITSIG  lpwork()
    3 100 FIFO TASK      RUNNING  init()
  261 105 FIFO TASK      WAITSEM  navigator()
  134 250 FIFO TASK      WAITSEM  sensors()
  199 177 FIFO TASK      WAITSIG  sdlog2(20016050, 20016060, 20015620, 20015630, 20015640, 20015650, 20015660, 20015670)
  136 140 FIFO TASK      WAITSIG  commander(10004330, 1000fff0)
  105 220 FIFO TASK      WAITSEM  gps(10004b40)
  138  50 FIFO PTHREAD   WAITSEM  commander_low_prio(0)
  107  90 FIFO TASK      WAITSEM  dataman()
  143 240 FIFO TASK      WAITSEM  px4io()
  240 250 FIFO TASK      WAITSEM  attitude_estimator_q()
  242 250 FIFO TASK      WAITSEM  lp_estimator()
  149 100 FIFO TASK      WAITSIG  mavlink_if0(20008330, 1000c350, 1000c9e0, 10003fd0)
  150 175 FIFO PTHREAD   WAITSEM  mavlink_rcv_if0(2000e560)
  248 250 FIFO TASK      WAITSEM  mc_att_control()
  186 100 FIFO TASK      WAITSIG  mavlink_if2(200170f0, 20017100, 20017110, 20017120, 20017130, 20017140, 20017160, 2001)
  187 175 FIFO PTHREAD   WAITSEM  mavlink_rcv_if2(20017190)
  252 250 FIFO TASK      WAITSEM  mc_pos_control()
  157 100 FIFO TASK      WAITSIG  mavlink_if1(20012e20, 20012e30, 20012e40, 20012e50, 20012e70, 20012e80, 20012e90, 2001)
  158 175 FIFO PTHREAD   WAITSEM  mavlink_rcv_if1(20012ed0)
nsh> sd_bench
nsh: sd_bench: command not found
nsh> commander check
WARN  [commander] PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
WARN  [commander] Warning: Primary gyro not found
WARN  [commander] Preflight check: FAILED
WARN  [commander] PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
WARN  [commander] Warning: Primary gyro not found
WARN  [commander] Prearm check: FAILED
```

sanderux, 2016-08-09:
I have seen this behavior too

art-mx, 2016-08-10:
Same thing. Is Master usable with this?

sanderux, 2016-08-15:
This is still happening. It makes updating very annoying as a sensor recalibration is needed.
Restoring params (through qgc) doesnt help against this.

LorenzMeier, 2016-12-26:
I'm pretty sure this were stale params and / or power cycle during changes. I never loose my params.

closed 2016-12-26 by LorenzMeier