PX4/PX4-Autopilot #5277: MAVLink-UAVCAN bridge is not working properly

pavel-kirienko, 2016-08-08:
See https://github.com/PX4/Firmware/pull/5271#issuecomment-238280605

@mhkabir @bendyer

pavel-kirienko, 2016-08-08:
What needs to be done so far:

1. Get Lorenz enable the support for setting the source component ID for outgoing MAVLink messages (https://github.com/PX4/Firmware/pull/5271#issuecomment-238368758).
2. Route `uavcan.protocol.NodeStatus` messages via the MAVLink-UAVCAN bridge, transforming them into MAVLink heartbeat messages.

LorenzMeier, 2016-11-13:
@DonLakeFlyer Current state with UAVCAN enabled:

[image]

LorenzMeier, 2016-11-13:
@mhkabir @pavel-kirienko Can you re-test on master?

pavel-kirienko, 2016-11-15:
There were reports of `Dynamic ID / Update` being silently reset to `Motors / Update`, which I couldn't reproduce. Can anyone confirm that?

Re testing, I should probably be able to do it later.

LorenzMeier, 2016-11-15:
@pavel-kirienko I can't see how this should be happening. But I did clean up the enable params here: https://github.com/PX4/Firmware/pull/5866

LorenzMeier, 2016-11-16:
Seems to be all fixed.

closed 2016-11-16 by LorenzMeier