PX4/PX4-Autopilot #5286: hdop is not eph

LONELY-WOLF, 2016-08-09:
Mavlink now reports hdop as eph and vdop as epv.
Here: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L1080
It's not useful if you have precise GPS receiver. Our Ashtech GPS gives us 0 or 1. Before those changes it reported EPH/EPV in cm.

LorenzMeier, 2016-08-09:
We did that to be consistent with existing deployment in the field. Please feel free to propose an extended MAVLink message with these fields added, but in the scope of the existing message definition it will stay as is.

closed 2016-08-09 by LorenzMeier

sanderux, 2016-08-09:
@LONELY-WOLF how are you connecting your Ashtech receiver?

kd0aij, 2016-08-09:
Is it correct that right now, QGC displays hdop and vdop, but LPE requires
eph < 3.0 (LPE_EPH_MAX)?
Is it possible to see eph and epv using QGC?