PX4/PX4-Autopilot #5290: Flight mode / throttle inspection

LorenzMeier, 2016-08-10:
@julianoes This user reported the vehicle rise uncontrolled. Not sure yet what caused it.
http://logs.uaventure.com/view/x2pFutLTCB7QdTTuBgRZJL

tumbili, 2016-08-10:
@LorenzMeier @julianoes @jgoppert  Check estimator innovations (they are horrible) and the reported gps speed (over 100 m/s in the vertical direction). GPS did indicate that the values are bad (eph and epv went up very high). The question is if the estimator (LPE) did a good job in rejecting the measurements. The vertical velocity seemed to pretty much follow the gps measurements.

tumbili, 2016-08-10:
@jgoppert Maybe you could have a closer look at this log?

jgoppert, 2016-08-10:
It looks like GPS raw data  decreased 500 m suddenly.

sanderux, 2017-04-18:
@jgoppert can you do something with this or do we close it as failing hardware?

jgoppert, 2017-04-18:
Probably best to close it.

closed 2017-04-19 by mhkabir

ecmnet, 2017-04-19:
@jgoppert Just a remark from my side. I had some similar issues regarding LPE's Z estimation relying on LIDAR only. It was in all cases the same reason. LIDAR estimation faulted, but did not recover (at least fast enough) and my quad crashed into the ceiling. I don't know, why the LIDAR estimation did not recover, as the raw LIDAR measurements were ok, but I noticed e.g. some dependencies on proper initialization (e.g. inactivated land estimation).