PX4/PX4-Autopilot #5298: Work Flow flexibility

davids5, 2016-08-11:
@dagar, @jgoppert FYI: @LorenzMeier 

Lorenz asked me to ask you both for a solution to this:

Discussion: https://github.com/PX4/Firmware/pull/5107#issuecomment-233989861

> I need to have targets built that are not going to be deployed (yet) via qgc and for now I do not want them in the package but need them built for verification.
> 
>... 
>  
> How can we implement super low maintenance way to add targets in a commit that need to be build but do not case merge thrashing?
> 
> -include (additional_targets.mk) ?

We need a solution where:

We do have to modify and merge/unmerge the MakeFile and .travis.yaml constantly.

The original build system had a feature that a build target added to the config directory that had, I think, _default - got built.  - not workable for what er do now, but it sure was a low overhead and diff noise solution.

My simple minded Make based solution would be to add a  file additional_targets.mk, that never get committed to master or is empty on master.  Then use the minus include option in the Makefile. 

Would you both please put your heads together and suggest a solution, and commit it to master.

dagar, 2016-08-11:
Here's the current target setup
qgc_firmware: builds on travis-ci and uploads to s3 for qgc flashing
check: builds all targets on semaphore-ci and runs testing, no upload

What about adding them to check single line? Would that reduce the merging pain?

davids5, 2016-08-12:
I already did that. It would be avoided with an include bases solution.

jgoppert, 2016-08-12:
CMake just needs the correct arguments to be passed, -DCONFIG="your_extra.config" So we are really just talking about how to easily alias target lists. Why not use a python helper script. It is much easier to maintain than a Makefile and we could have lists, usage etc.

davids5, 2016-08-12:
@dagar @jgoppert, FYI:@LorenzMeier 

If that makes the most sense, @jgoppert would you put is a PR with a solution.

dagar, 2016-08-15:
@davids5 I'm happy to implement additional_targets.mk if that would be helpful for you.

davids5, 2016-08-15:
Please :)

dagar, 2016-11-08:
Resolved? Please reopen if that's not that case.

closed 2016-11-08 by dagar