PX4/PX4-Autopilot #5303: After BSP and Linux updates Snapdragon px4 start still has problem

VictorSu2, 2016-08-12:
Hi,
After doing the BSP and Linux image update, the errors shown are less.
But still remains error such that px4 doesn’t work properly.
The error message is as shown:

> root@linaro-developer:/home/linaro# ls
> FTM_AP dataman mainapp.config px4
> root@linaro-developer:/home/linaro# ./px4 mainapp.config
> sh: 1: cannot create /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor: Directory nonexistent
> ERROR [platforms__posix__px4_layer] krait update param 297 failed
> ERROR [platforms__posix__px4_layer] krait update param 646 failed
> | ___ \ \ \ / / / |
> | |/ / \ V / / /| |
> | _/ / \ / /| |
> | | / /^\ \ __ |
> _| \/ \/ |_/
> px4 starting.
> ERROR [muorb] Initialize Error calling the uorb fastrpc initalize method..
> ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
> INFO [sdlog2] [blackbox] /root/log/2016-08-03
> INFO [sdlog2] [blackbox] recording: 07_43_57.px4log
> INFO [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 103090 bytes
> INFO [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
> pxh> INFO [mavlink] using network interface wlan0, IP: 192.168.1.159
> INFO [mavlink] with broadcast IP: 192.168.1.255
> INFO [mavlink] partner IP: 192.168.1.155

Could anyone possibly figure out what problems it might be?

Thanks
Victor

julien10, 2016-08-29:
Any news on this subject?

julianoes, 2016-08-31:
@julien10 Sorry I'm currently quite busy. Will get back to this when I can, hopefully next week. Let me know if you find anything in the meantime.

VictorSu2, 2016-09-01:
@julien10 I have tried several DSP firmware and Linux image upgrade and have deleted the PX4 firmware and clone several times. But it still doesn't work. Now i am stuck in here......

julianoes, 2016-10-10:
@VictorSu2 have you ever resolved this? Could it be the same as #5612?

VictorSu2, 2016-10-11:
@julianoes I think i sovled this several days before. I ran the test program of DSP from https://github.com/PX4/dspal/tree/d517bb7841ae0fa5136cdfcace2d54b172418051 and i found out the tests all failed. Then i tried to upgrade ADSP and Linux by Ubuntu real machine instead of Virtual machine and the px4 start is finally clean. I think this note should be added to the px4 development's manual in the snapdragon section.

VictorSu2, 2016-10-11:
@julianoes I also find out some other problems that maybe caused by hardware issues. I got three snapdragon boards upgraded to same DSP firmware and linux image. Then ran the same px4 firmware. It turns out that the first board's accelerometer's calibration can't pass since the readings are wrong. When i put the board flat on the ground and the readings of accelerometer is -7 8 10(x, y, z) instead of 0 0 10(x, y, z). The second board's ESC port(J13) can't output uart signal while the first board does. Only the third board can output ESC signal and can pass the accelerometer's calibration. Do you think so?

julianoes, 2016-10-11:
@VictorSu2 Thanks for the feedback.

> When i put the board flat on the ground and the readings of accelerometer is -7 8 10(x, y, z) instead of 0 0 10(x, y, z)

We have seen the accelerometer on the Snapdragon board "fail" like this before (just with other values). Please contact Intrinsyc and ask for a replacement board.

> The second board's ESC port(J13) can't output uart signal while the first board does. 

If you followed the same install procedures, I don't see how this can be a software issue. Send it in to Intrinsyc as well ;).

> Then i tried to upgrade ADSP and Linux by Ubuntu real machine instead of Virtual machine and the px4 start is finally clean. I think this note should be added to the px4 development's manual in the snapdragon section.

I believe you but I don't understand why this would matter. If you can find differencies in the md5sums of the files that got copied or anything like that, it would be proof.

VictorSu2, 2016-10-11:
@julianoes thanks, i will contact intrinsyc for the hardware issues. 
Sorry i can't supply the proof that real machine works. 
I just tried the method and it worked.

julianoes, 2016-10-11:
Ok, I'll take note in the troubleshooting part.

closed 2016-10-11 by julianoes