PX4/PX4-Autopilot #5304: 1.4.1: mav_cmd_do_vtol_transition failed

jilinggui, 2016-08-12:
Hi ,guys .when i  test  automatic transition(mission, commanded)according to the page "[http://dev.px4.io/airframes-vtol-testing.html](http://dev.px4.io/airframes-vtol-testing.html)"under the sitl, when transition from MC to FW ,the vehicle  become  crazy,the  terminal output liake this :
`

> WARN [navigator] forwarding command 3000
> 
> WARN [vtol_att_contraol] failed setting min values
> INFO [lib__ecl] EFK baro hgt timeout - reset to baro
> INFO [tone_alarm] gps_warning
> WARN [lib__ecl] EFK GPS fusion timout - resetting to GPS
> WARN [lib__ecl] EFK GPS fusion timout - resetting to GPS
> INFO [platforms__posix__px4_layer] simulator is slow.Delay added:12016 us
> ...`

,What's the matter?

sanderux, 2016-08-12:
I am unable to duplicate this on master.
Did you try a make clean and update your gazebo?

jilinggui, 2016-08-12:
@sanderux Thanks a lot.now i am using gazebo6, I will update to gazebo7 and test again.

jilinggui, 2016-08-12:
@sanderux Hi sanderux,I have tried,but it still does not work.I work on VMware,i doubt that is the result of virtual machine is running pretty slow,do you have any good idea?

sanderux, 2016-08-12:
i have never seen gazebo work well on a VM, i would suggest you try it on a dedicated linux environment

dagar, 2016-08-12:
Try running gazebo without the gui.

    HEADLESS=1 make posix_sitl_default gazebo_plane

closed 2016-08-13 by sanderux

jilinggui, 2016-08-16:
@sanderux @dagar Hi,I have tested on a dedicated linux environment,and i also have test running gazebo without the gui, both can't normal to switch from MC to FW. Maybe  need to set some parameters?

sanderux, 2016-08-16:
Can you copy the console output when you run on a dedicated machine?