PX4/PX4-Autopilot #5312: Pixracer overheat

akru, 2016-08-13:
Month ago I buy china **pixracer** from [Aliexpress](http://www.aliexpress.com/item/Mini-Pixracer-Autopilot-xracer-V1-0-Version-FMU-V4-PX4-Flight-Controller-Board-for-QAV-Multicopter/32660706353.html). So it looks good with `ArduCopter-3.3.3`, but with latest versions of APM or PX4 firmware (in my opinion it's related) the PCB have overheat around main chip (50-70С for a long time).

Unfortunately I cannot have thermal imager for additional information. May be I shot whith it in the future.

kd0aij, 2016-08-13:
I doubt that the manufacturer of that board is reading these issues.

LorenzMeier, 2016-08-13:
Please contact the manufacturer. This is not a software issue.

closed 2016-08-13 by LorenzMeier