PX4/PX4-Autopilot #5314: 1.4.1 VTOL How to program transition switch?

tubeme, 2016-08-14:
Hi,

Just about to try the 1.4.1 and I cannot seem to find a way to program a transition switch.

In the release statement I see "VTOL: Dedicated transition switch for manual operation." But I dont see it in QGC. I use today's build of QGC.

AUX1 does not work anymore like in 1.3.4

Any ideas?

tubeme, 2016-08-14:
I must say 1.3.4 is ROCK SOLID! in all weather conditions. :) I must say BIG Thank you for the good work.

kd0aij, 2016-08-14:
The transition switch mapping was changed recently: https://github.com/PX4/Firmware/pull/5101
and I don't think QGC supports it yet. You'll have to manually assign the switch.

tubeme, 2016-08-14:
So how do i program the switch?

tubeme, 2016-08-14:
Thanks I changed it in the config.txt it works..

closed 2016-08-14 by tubeme