PX4/PX4-Autopilot #5316: v1.4.1 release, Pixracer, LPE rejects ALTCTL even with LPE_GPS_ON=0

kd0aij, 2016-08-14:


jgoppert, 2016-08-16:
Looking into this, any logs?

kd0aij, 2016-08-16:
No, I tested by connecting QGC via wireless telemetry (outdoors) and changing TX mode switch from manual to ALTCTL. QGC reported that the mode switch was rejected, so I didn't even arm.

kd0aij, 2016-08-28:
this looks OK on latest master

closed 2016-08-28 by kd0aij

saeidkazemi, 2017-02-11:
hi
We have some unknown problems to compile your code and we would be pleased if you could help us to fix it.
I have dealt with the problem of Position Control Mode actually the system rejects this mode.
We do not have GPS and want to run the Position Control with optical Flow Sensor.
We cannot find the LPE-GPS-on in Qground to turn it off, so how can we turn off the GPS?
We set the sys_group in LPE mode but again it rejects the position control mode. Do you know the reason?

Pixhawk version 2

jgoppert, 2017-02-12:
Please set sys_mc_est_group to lpe and reboot.

saeidkazemi, 2017-02-12:
actually we did tat but again we had the same problem
our Firmware link is:
https://github.com/mhkabir/Firmware/

and also our flight controller is pixhawk 2

mhkabir, 2017-02-12:
Why are you using my fork instead of master?

saeidkazemi, 2017-02-12:
what do you mean by fork exactly?

mhkabir, 2017-02-12:
Why are you not using https://github.com/PX4/Firmware/ ?

saeidkazemi, 2017-02-12:
actually we used that firmware and we set sys_mc_est_group in LPE mode then we reboot the pixhawk and again we didn't have LPE-GPS-ON

jgoppert, 2017-02-12:
If you are using px4fmu-v2 make sure you use : make px4fmu-v2_lpe upload