PX4/PX4-Autopilot #5328: Restore params needs multiple runs

sanderux, 2016-08-16:
As some params are optional and only load when others are set, restoring params from a backup after a reset or on a new device requires multiple runs.

Is there any way around this?

LorenzMeier, 2016-08-16:
Yes there is. The PX4 param system does not forbid to try to write a "non-existent" parameter. The QGC side might try to be too smart about it when it loads the file.

I guess what's needed is a try-and-report-failure approach rather than attempting to prevent the user from the start.

DonLakeFlyer, 2016-08-16:
Here is what happens if QGC sends params it does not know about:
1) User has a file with params currently not know to QGC, say calibration settings
2) User loads that file and sends to vehicle
3) QGC still shows various calibration state as incomplete
4) User goes to look for calibration params in parameter list and they are not there

There is no way for QGC to deal with 3 and 4 since known param set only updates at vehicle boot time. QGC would need to update known param set on the fly instead of from initial parameter load. That is going to get complicated really fast. Since an unknown param won't really turn into a known param unless it gets a write ack back from the vehicle. I think it's doable but a heck of a lot of work and additional complexity to the parameter protocol for optional params. I still come back to why can't we toss optional params now that we have cache hit support.

sanderux, 2016-08-16:
How about a qgc reboot after a param restore?

DonLakeFlyer, 2016-08-18:
That would work if QGC just sends all param across. I need to look at whether the param code will have a problem with sending params it doesn't know about. It will likely cause problems if the params really are no longer there as opposed to an optional param problem. Unfortunately making it work flawlessly will require a re-arch of the param loader protocol.