PX4/PX4-Autopilot #5330: LPE_GPS_ON parameter always resets to default on restart

agribov, 2016-08-16:
Setting this parameter to "disabled", then restarting the PixHawk board, causes the parameter to reset back to "enabled". Also tried setting this parameter and another one at the same time, then restarting, and this one reset to default while the other held it's new value.

It doesn't seem to take effect before the restart either -- Copter will always reject altitude control, even right after setting the parameter before restarting.

kd0aij, 2016-08-16:
@jgoppert sounds like this is the root cause for https://github.com/PX4/Firmware/issues/5316

agribov, 2016-08-16:
Yea, rejected alt hold is what led me to this issue as well -- This parameter also doesn't have a "restart required" dialog, so on first glance it looks like the parameter is successfully set and doesn't depend on reboot, when in reality it doesn't change and a reboot exposes that.

jgoppert, 2016-08-16:
I agree it should have reboot required flag, odd that it is resetting through, I haven't seen that on my end yet. It might be a recent issue with master?

agribov, 2016-08-17:
Maybe, I believe that I'm running 1.4.1 dev, and can check for sure in the morning. Is there anything I can upload or a test I can run that would help identify the issue?

agribov, 2016-08-18:
Any word on this? @LorenzMeier , do you know who could address this? Seems like a simple issue, but there's no way to do Optic Flow flights without GPS until it's resolved.

jgoppert, 2016-08-18:
Actually, if you set LPE_GPS_ON = 0, it should give you a lock without reboot. I hopefully will be able to fly soon, but I've checked it in sitl on master and it works.

kd0aij, 2016-08-18:
as of today, beta exhibits the problem. Changing LPE_GPS_ON to zero had no
effect before (or after) reboot.
Wasn't able to fly master because, in addition to losing the primary gyro
on reflashing, the gyro cal process hangs. (pixracer)
I flew beta with the gps unplugged (after compass recal) and altctl mode
was still rejected. But PID gain scheduling (AKA TPA) worked well :)

LorenzMeier, 2016-08-18:
I can't reproduce this on Pixhawk. Where did you reproduce this?

LorenzMeier, 2016-08-18:
I can confirm the gyro cal issue. Odd.

agribov, 2016-08-18:
I got this while running 1.4.1dev on a genuine Pixhawk 2.4.8.

By gyro cal issue are you reffering to our discussion from the discuss page? My issue seems to have goneaway on one Pixhawk just by recalibrating the gyro, still haven't checked back on the RTFhawk I originally tried with.

LorenzMeier, 2016-08-19:
I've fixed the gyro cal issue in master / beta as well. I'm closing this since testing suggests this was something local.

agribov, 2016-08-19:
What makes you think this is local? Both @kd0aij and I are experiencing it separately. Maybe there's another parameter or root behind the issue that's set on our device, and not yours?

LorenzMeier, 2016-08-19:
I missed you had two repro instances. I experienced something similar with an LPE parameter today on a Pixracer. I'm wondering if its tied to the C++ param system it uses or just a general regression.

gaochangyu, 2016-11-02:
I've just encountered this problem with my pixhawk 2.4.8 hardware just now. It is the same problem @agribov described. Does anyone already have a solution for this?
I tried v1.4.4 and v1.5.0(build from source). LPE_GPS_ON is always reset to ENABLED when restarted.

davids5, 2016-11-02:
@gaochangyu 
See 
http://diydrones.com/group/pixhawk/forum/topics/pixhawk-2-4-8-from-bangood-not-working-as-expected?commentId=705844%3AComment%3A2244045&groupId=705844%3AGroup%3A1582361

gaochangyu, 2016-11-03:
@davids5 Do you mean I should try reset all the parameters?

davids5, 2016-11-03:
yes

gaochangyu, 2016-11-03:
I did a experiment and had some findings. I believe this is a bug in parameter writing.
I did the experiment with a genuine 3dr px4 2.4.6 and qgroundcontrol 3.0.0 on linux.
Here is the work around.
1. change the LPE_GPS_ON to disabled;
2. change another parameter;
3.wait for several seconds LOL;
4.and then reboot the board, you'll find LPE_GPS_ON is successfully saved.

Well, it sounds crazy but it really worked for me. maybe you can look into this? @LorenzMeier

iZhangHui, 2016-11-03:
@gaochangyu Could you verify my PR #5797  working for you?

gaochangyu, 2016-11-03:
@iZhangHui Yep, It worked.

closed 2016-12-23 by jgoppert

saeidkazemi, 2017-02-12:
hi
We have some unknown problems to compile your code and we would be pleased if you could help us to fix it.
I have dealt with the problem of Position Control Mode actually the system rejects this mode.
We do not have GPS and want to run the Position Control with optical Flow Sensor.
We cannot find the LPE-GPS-on in Qground to turn it off, so how can we turn off the GPS?
We set the sys_group in LPE mode but again it rejects the position control mode. Do you know the reason?

Pixhawk version 2