PX4/PX4-Autopilot #5336: Receiving a custom mavlink message

ndepal, 2016-08-17:
I want to receive a custom mavlink message in `mavlink_receiver.cpp`. I have added my message definition in `common.xml`.

Before generating the mavlink header files, I see that I am receiving the new mavlink messages (which I am sending from QGC) by printing the incoming message IDs in [`handle_message()`](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L162).

In order to decode the mavlink message I generate the header files for both v1.0 and v2.0 (I have added my new message definition in both versions of `common.xml`) using `mavgenerate.py` with the appropriate paths and version selected.

After that I no longer see the incoming message ID of my custom message being printed. `handle_message()` does not get called anymore for that message because [`mavlink_parse_char()`](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L2172) never returns true.
It appears to be the fault of the v2.0 headers. Removing my message from v2.0 `common.xml` and generating that code once again allows `handle_message()` to be called for my message but I can't decode it because the compiler wants the headers to be in v2.0.

What might I be doing wrong?

mhkabir, 2016-08-17:
I'm running into this as well. I've been syncing our stable branches with PX4 upstream for the last few weeks, and custom messages don't quite seem to be working OK. Similar results as @ndepal .

ndepal, 2016-08-17:
Is there a way to go back to mavlink v1.0 in the firmware?

ndepal, 2016-08-17:
I don't get what identifies my message as a custom one. I define it in `common.xml`, along with other standard messages and give it an id of 145, which is not in the range of 'extensions', as specified [here](http://qgroundcontrol.org/mavlink/create_new_mavlink_message).  As far as I can tell, there is no discernible difference between my new message and existing ones such as `HIGHRES_IMU`.

closed 2016-12-28 by LorenzMeier

aysh-shariff, 2017-01-20:
@ndepal Could you fix this issue ? I am facing a similar problem and could do with some help.

ndepal, 2017-01-20:
@aysh-shariff I have not fixed this issue. At the time when we ran into this, we worked out a workaround which we are still using. This is because we have frozen our PX4 version from upstream at that point.
If you are still experiencing this issue you should re-open this issue or create a new one.