PX4/PX4-Autopilot #5338: Unplanned climb in mission after backward transition

mswingtra, 2016-08-18:
On Tuesday, during one of the flights, after backward transition aircraft started to climb without a reason as the setpoint was below it. You can see in the log that the climb has been commanded by the multicopter controller, but its velocity setpoints should be based on waypoints, and the altitude of the active setpoint was below current altitude.
Mission was set up on the Android tablet, with QGC https://github.com/mavlink/qgroundcontrol/commit/948fcd472fe5da65d1ba287795533648da49216e. 
I have run the same mission in SITL, but I could not reproduce this behaviour.

Has anyone encountered similar behaviour? 

Mission: [mission_from_tablet.txt](https://github.com/PX4/Firmware/files/424374/mission_from_tablet.txt)
Log: http://logs.uaventure.com/view/MVtuASWyYPu8hZo8AykLMJ
Flown with master 5712771.

AndreasAntener, 2016-10-18:
That could have been #5301

mswingtra, 2016-12-14:
I agree that could have been an issue here. It has also never occurred again, so I'm closing here.

closed 2016-12-14 by mswingtra