PX4/PX4-Autopilot #5348: Build error occured 'cause of a warning. (make posix_sitl_default jmavsim)

junhohan, 2016-08-19:
I got an error when run below.

    $ make posix_sitl_default jmavsim

The error is like this.

> ../src/modules/systemlib/print_load_posix.c: In function ‘print_load’:
> ../src/modules/systemlib/print_load_posix.c:117:20: error: variable ‘basic_info’ set but not used [-Werror=unused-but-set-variable]
>   task_basic_info_t basic_info;
>                     ^~~~~~~~~~
> compilation terminated due to -Wfatal-errors.
> cc1: all warnings being treated as errors
> [15/86] Building CXX object src/lib/te...estimation.dir/terrain_estimator.cpp.o
> ninja: build stopped: subcommand failed.
> make: *** [posix_sitl_default] Error 1

There is a varable not used in "print_load_posix.c".

    task_basic_info_t basic_info
    basic_info = (task_basic_info_t)tinfo;

I think this should be deleted under the "-Werror=unused-variable"  option.

bkueng, 2016-08-23:
yes, you're right. see #5364

closed 2016-08-23 by bkueng

Unity-W, 2019-07-12:
i meet the same problem,but
where is the "I think this should be deleted under the "-Werror=unused-variable" option."
I can't find it