PX4/PX4-Autopilot #5353: MAVLink INTERVAL change with mavlink_message

lefos99, 2016-08-19:
#4794 

I wanted to do the same thing with @sirPerna  but so far I fail.
First of all, I create successfully a MAVConnInterface

```
catch (mavconn::DeviceError &ex)
        {
            try
            {
                url = "serial:///dev/ttyACM0";
                link = mavconn::MAVConnInterface::open_url(url, 1, 1);
            }
            catch (mavconn::DeviceError &ex)
            {
                ROS_FATAL("OSD: %s", ex.what());
                ros::shutdown();
                return;
            }
        }
```

And then I try to send a COMMAND_LONG message like that:
```
        mavlink_command_long_t change_rate_command = { 0 };
        change_rate_command.target_system    = 1;
        change_rate_command.target_component = 1;
        change_rate_command.command          = MAV_CMD_SET_MESSAGE_INTERVAL;
        change_rate_command.confirmation     =  0;

        change_rate_command.param1            =  MAVLINK_MSG_ID_SYSTEM_TIME;
        change_rate_command.param2            =  3000000.0; // microseconds
        change_rate_command.param3            =  3000000.0; // microseconds

        mavlink_message_t ratio_changer_msg3;
        mavlink_msg_command_long_encode(50, 1, &ratio_changer_msg3, &change_rate_command);
        link->send_message(&ratio_changer_msg3, 1, 1);
```
I also tried:
```
mavlink_message_t ratio_changer_msg;
mavlink_msg_message_interval_pack(1,50,&ratio_changer_msg,MAV_CMD_SET_MESSAGE_INTERVAL,-1);
link->send_message(&ratio_changer_msg, 1, 0);
```
and
```
mavlink_message_t ratio_changer_msg2;
mavlink_msg_command_long_pack(1,50,&ratio_changer_msg2,1,1,MAV_CMD_SET_MESSAGE_INTERVAL,true,MAVLINK_MSG_ID_SYSTEM_TIME,-1,-1,-1,-1,-1,-1);
link->send_message(&ratio_changer_msg2, 1, 1);
```

but nothing seem to work! 
Any help??

sirPerna, 2016-08-21:
Seems like you don't use the same sys and com id when you write the change_rate_command frame and when you actually pack it to ratio_changer_msg. I recommend to get these ids from the heartbeat message so you are sure you going to configure the correct device.

lefos99, 2016-08-22:
I got :
> sysid: 1
compid: 1

Well when I pack I use 
* @param system_id ID of **THIS** system
* @param component_id ID of **THIS** component (e.g. 200 for IMU)
But even when I use the same ones, doesn't seem to work...
Any other ideas?

sirPerna, 2016-08-22:
Check mavlink_msg_command_long_encode. You pack the message with sysid 1 and compid 50. Try to use the same ids you used here:
change_rate_command.target_system    = 1;
change_rate_command.target_component = 1;

lefos99, 2016-08-22:
Yes I did what you said and now the code is:
```

        mavlink_command_long_t change_rate_command = { 0 };
        change_rate_command.target_system    = 1;
        change_rate_command.target_component = 1;
        change_rate_command.command          = MAV_CMD_SET_MESSAGE_INTERVAL;
        change_rate_command.confirmation     =  0;

        change_rate_command.param1            =  MAVLINK_MSG_ID_SYSTEM_TIME;
        change_rate_command.param2            =  -1; // microseconds
        change_rate_command.param3            =  -1; // microseconds

        mavlink_message_t ratio_changer_msg3;
        mavlink_msg_command_long_encode(1, 1, &ratio_changer_msg3, &change_rate_command);
        osd_link->send_message(&ratio_changer_msg3, 1, 1);
```

But the rate doesn't change! (I putted now -1 for rate in order to stop the stream....)

sirPerna, 2016-08-22:
@LorenzMeier Maybe you can help here?

closed 2016-08-30 by lefos99

omegix, 2019-03-20:
@lefos99 were you ever able to solve this issue?  I am running into the same thing two years later.