PX4/PX4-Autopilot #5354: SITL large setpoint deviations

annesteenbeek, 2016-08-19:
I am currently developing a precision autoland system controlled by mavros using Offboard control and running all my tests in the gazebo simulator. 
But when using the mavros position control, the quadcopter moves around the setpoint quite a bit, and never stays still. I know this is normal for real quadcopters (to some extent) but in the simulation, without external influences like wind, the copter should remain almost perfectly still (except for simulated sensor inaccuracy I would assume) and not move tens of centimeters.
The same for velocity control, using a simple proportional controller to control the position using the mavros local position as feedback, I theoretically should be able to achieve near 0 error in the simulator but the error keeps fluctuating. And when i apply 0 z velocity, the quad moves up instead of doing nothing. (however I read in a previous Issue this could be due to wrong barometer parameter, though I haven't tested this yet).

Is there something wrong with the default SITL parameters causing the copter to behave so unstable? Or is there something else I am missing like a large amount of noise being used for testing the sensors? Are there parameters to control this behavior, there was nothing I could find about this in the documentation.

jgoppert, 2016-08-19:
There is simulated GPS drift. If you use optical flow it will be slightly less. What estimator are you using? If you are doing gazebo simulation, I would suggest this patch: https://github.com/PX4/Firmware/pull/5255

Should make your life easier. Hopefully we will get it merged soon.

annesteenbeek, 2016-12-17:
Knowing where it came from helped, thanks, sorry for closing this so late.

closed 2016-12-17 by annesteenbeek