PX4/PX4-Autopilot #5358: LPE: Does not accept msg_vision_speed_estimate messages as vision input

ecmnet, 2016-08-21:
Currently, LPE accepts both, MOCAP and VISION messages for position. However input for speed estimates are not supported.

mhkabir, 2016-08-21:
Indeed. It's trivial to add, and was even in there at some point, but no one tested/maintained it, so it was removed.

ecmnet, 2016-08-22:
Trying to work on this, I found out that the uORB message vision_position_estimate differs from the MAVLINK definition. The uORB message contains velocity, while the mavlink message does not. On the other hand, there is another mavlink message called msg_vision_speed_estimate. My question now: What is the best approach, either
- Mapping msg_vision_speed_estimate to the corresponding vision_position_estimate fields, or
- Creating a new uORB-message according to the MAVLink message msg_vision_speed_estimate, or
- Extending the MAVLink message msg_vision_position_estimate according to its uORB correspondence.

Any thoughts?

jgoppert, 2016-08-22:
What is your use case exactly. Are you doing optical flow or 3D visual odometry without a direct position estimate? My original thought was that on these systems they should just send the integrated position instead of the velocity because that would be more accurate and won't drift over time much as long as the vision data is good. The flow code does the same thing, and when the flow quality degrades it shuts off and then will reinitialized when quality returns. The alternative where you use the vision data as a velocity measurement means all other velocity measurement sources (accel) etc. go into the integration and even though you may be tracking the same feature for 10 min visually you will drift considerably. But I think supporting both would be a good thing so we can compare. That would probably mean having a separate velocity only vision correction function, I don't ever see a use case with both position and velocity, and that will take more cpu since it is 6x6 instead of 3x3.

ecmnet, 2016-08-22:
I am currently experimenting with an onboard R200 implementation. Although its speed estimation seems to be quite ok, I face currently issues integrating speeds there. My assumption was that feeding LPE with speeds could improve poshold accuracy.

jgoppert, 2016-08-22:
How good is the velocity data, what library are you using? I've been experimentin with that as well.

ecmnet, 2016-08-22:
I am using boofcv, a Java based library. Meanwhile I added a propere Position estimation to it. Looks quite promising so far. Rotation and re-initialization are currently my main tropics. What was your approach?

jgoppert, 2016-08-22:
@ecmnet I was using opencv feature tracking and then calculating 3d feature position with depth information. Didn't really yield robust results. The vision based correctoin is just based on the rate you send it, so shouldn't be limited.

ecmnet, 2016-08-23:
@jgoppert I'll keep you updated and share logs as soon I get some reasonable results

ecmnet, 2016-08-23:
@jgoppert After fixing the rotation issue, here some results comparing vision based position (yellow) vs. LPE (with flow, VISION set to Off, red gaph), vehicle moved around manually:

[image]

One of my problems now is when trying to fuse vision with flow, that flow position has driftet away during (simulated) landing (That is what you see in the graph also) Fusing with reasonable weights results in a mess. Increasing flow std.dev.  seems not to be the solution as the flow impact is too small then. How do you reset flow estimation in case of faults? Is it set to the current estimation?

ecmnet, 2016-08-23:
Another comparison (same method, max features increased to 100) shows the flow-drift when simulating a takeoff:

[image]

ecmnet, 2016-08-23:
Tried to fuse both (red now fused vision with flow):

[image]

jgoppert, 2016-08-23:
This issue is alignment between the flow and vision. If we treat them as position measurements we end up with this issue. We can solve it by adding alignment states, but adding alignment states for each type of measurement seems cumbersome. What I tried to do was estimate the alignment offset at initialization (vision_home etc.), we should look into why that isn't working well.

ecmnet, 2016-08-23:
Worth to mention, that I initialize raw vision position with the latest published local position. This also happens, when a vision fault is detected. Maybe the issue is related to the different point in time during initialization: Vision is available on ground, while flow starts at a certain distance above ground. The drift might occur in between those two points in time. 
I am not very deep into LPE maths, but could it be that it is not clear, what the leading position is, and which of them should be corrected via an offset?

jgoppert, 2017-04-28:
no plans to support this currently, will probably be added to IEKF.

closed 2017-04-28 by jgoppert