PX4/PX4-Autopilot #5363: px4/SITL/gazebo cmake error

csorice147, 2016-08-22:
This cmake error is not very descriptive - was wondering if anyone has suggestions
I'm using ros indigo & ubuntu 14.04, fyi

`[100%] Built target px4
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: gazebo
model: none
build_path: /home/csorice/px4/src/Firmware/build_posix_sitl_default
empty model, setting iris as default
make[5]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[6]: *** No targets specified and no makefile found.  Stop.
make[5]: *** [gazebo_build] Error 2
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 2
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [posix_sitl_default] Error 2
`

LorenzMeier, 2016-08-24:
Can you run cmake in verbose mode and see what it says?

LorenzMeier, 2016-08-24:
Please try again with an updated master.

yota-code, 2016-08-25:
Hello, I think I had the same problem. I updated to master this morning and I still got an error. How can I set cmake to verbose to help you unravel this issue ? Here is the last errors I get after a clean make : 

```
Scanning dependencies of target gazebo
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: gazebo
model: none
build_path: ~/forge/Firmware/build_posix_sitl_default
empty model, setting iris as default
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
-- install-prefix: /usr
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so  
-- Boost version: 1.58.0
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- catkin DISABLED
-- Checking for module 'protobuf'
--   Found protobuf, version 2.6.1
Gazebo version: 7.3
-- Checking for module 'eigen3'
--   No package 'eigen3' found
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  CMakeLists.txt:70 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "~/forge/Firmware/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "~/forge/Firmware/build_gazebo/CMakeFiles/CMakeError.log".
Makefile:240: recipe for target 'gazebo_build' failed
make[5]: *** [gazebo_build] Error 1
src/firmware/posix/CMakeFiles/gazebo.dir/build.make:57: recipe for target 'src/firmware/posix/CMakeFiles/gazebo' failed
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 2
CMakeFiles/Makefile2:9689: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/all' failed
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
CMakeFiles/Makefile2:9696: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:3419: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:172: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 2

```

julianoes, 2016-08-25:
```
-- Checking for module 'eigen3'
--   No package 'eigen3' found
```

LorenzMeier, 2016-08-25:
@yota-code You did not follow our setup instructions. Please go back to the installation instructions on http://dev.px4.io

yota-code, 2016-08-25:
Indeed, I went too fast, coffee was not strong enough this morning x_x
I missed a few package (like protobuf and libeigen3...) sorry for the noise :)

julianoes, 2016-08-25:
No worries, cool. If something is missing in the [Devguide](https://github.com/PX4/Devguide), let us know.

closed 2016-08-25 by julianoes