PX4/PX4-Autopilot #5374: error: ‘HEX_T’ is not a member of ‘MultirotorGeometry’

yota-code, 2016-08-24:
During compilation (either `make posix_sitl_default jmavsim` or `make posix_sitl_default gazebo`) I got this error.

OS is ubuntu 16.04
Git revision of the source : `git rev-parse master                   
fadac79617055eb9db705aabea4cce582967bbe9
`

```
Firmware/src/modules/systemlib/mixer/mixer_multirotor.cpp: In static member function ‘static MultirotorMixer* MultirotorMixer::from_text(Mixer::ControlCallback, uintptr_t, const char*, unsigned int&)’:
Firmware/src/modules/systemlib/mixer/mixer_multirotor.cpp:175:14: error: ‘HEX_T’ is not a member of ‘MultirotorGeometry’
   geometry = MultirotorGeometry::HEX_T;
              ^
compilation terminated due to -Wfatal-errors.
```

julianoes, 2016-08-24:
Search for the file: `mixer_multirotor.generated.h` and delete it.

Or alternatively do `git clean -d -f -x` but make sure with `git clean -d -f -x -n` first that it doesn't delete anything that you need.

yota-code, 2016-08-24:
My local repo was clean, but indeed a `make clean` removed previously generated files and did the trick :)

Thanks for the amazing reactivity. Now I'm stuck on the #5363 issue, which is much better (or worse, dunno yet :D)

closed 2016-08-24 by yota-code