PX4/PX4-Autopilot #5383: UORB Message generation CMake msg_file_names does not really do anything

mazahner, 2016-08-25:
I've noticed that when adding or removing message definitions from
Firmware/msg/CMakeLists.txt it does not really matter of what is compiled in, in the best case it just spits out compilation errors. (the px4_base.cmake file only considers the -d option for complete directories which is the root of the evil, because it's not consistently used)

I have a solution working for me which does only compile the files listed in the CMakeLists.txt which is fine. I'm wondering: is there any intention to actually use the message generation feature more selective? and only apply it to what i actually want to compile based on the CMakeLists?

Thanks, Markus

bkueng, 2016-08-25:
The reason for the CMakeLists.txt is build dependency: if we add or remove a message, cmake handles the new/removed dependencies properly (incremental rebuild), without the list, cmake has no way of knowing that.

But you are right, there is an inconsistency in that the generator uses all msg files in the dir. It never really bothered us, because when we add a message, we always add it to the cmake as well.
If you open a PR to make the generator use only the msg files from the CMakeLists.txt, we'll gladly accept that.

mazahner, 2016-08-25:
PR #5384

I'm adding the Firmware/msg folder as a default inclusion path.
If you were to have message definitions in other (than Firmware/msg)  folders and you wanted refer to these messages, it will likely not work. (wouldn't have worked before either)

closed 2016-08-26 by mazahner