PX4/PX4-Autopilot #5391: Snapdragon Flight upload error: "waiting for bootloader..."

julien10, 2016-08-26:
I'm trying to upload the Hello Sky example app on the Snapdragon Flight, following the guide at http://dev.px4.io/tutorial-hello-sky.html
When I try to make upload, it gets stuck at:
`Loaded firmware for 9,0, size: 1025348 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.`

I have tried to unplug and replug, and also `reboot2fastboot`, but it did not change anything.
Is there something different with the Snapdragon Flight?

julianoes, 2016-08-26:
Are you following these steps to upload it?
http://dev.px4.io/starting-building.html#qurt--snapdragon-based-boards

It seems like you are trying to upload something to a Pixhawk/Pixracer from the output.

julien10, 2016-08-26:
No, I was using the Pixhawk commands, I did not realise there were different, my bad! 
So I just have to build and then upload using `make eagle_default upload`, or do I also have to modify a cmake file (as they show in the tutorial)?

julianoes, 2016-08-26:
No, you just use what's in dev.px4.io. Which tutorial do you mean?

closed 2016-08-27 by LorenzMeier

julien10, 2016-08-27:
http://dev.px4.io/tutorial-hello-sky.html#step-3-register-the-application-in-nuttshell-and-build-it