PX4/PX4-Autopilot #5393: px4 cmake error with gazebo

csorice147, 2016-08-26:
ubuntu 14.04 // gazebo 7.3.1 

I downloaded gazebo7 as specified here: http://gazebosim.org/tutorials?tut=ros_wrapper_versions#Gazebo7.xseries 

but then when I try to do `make posix_sitl_default gazebo`

I get this error:
`[  1%] [  2%] CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.
`

LorenzMeier, 2016-08-27:
We're not aware that our install instructions wouldn't work. I can re-check later on my machine.

antoinealb, 2016-09-20:
Installing `libgazebo7-dev` fixed the problem for me. I also needed to install `libopencv-dev` and `protobuf-compiler` to be able to build the simulation.

julianoes, 2016-09-30:
Thanks @antoinealb for the comment. `protobuf-compiler` and `libopencv-dev` are already in the list of packages that need to be installed, and I've added a comment that `libgazebo7-dev` is needed in https://github.com/PX4/Devguide/pull/44.

closed 2016-09-30 by julianoes

bigbellmercy, 2021-03-15:
FYI, I report the same error during a ROS package is built.
After `libgazebo9-dev` installed by  `sudo apt install libgazebo9-dev` command, it disappeared.

### Environment
Ubuntu 18.04 in x86
ROS Melodic, 'catkin build' environment
PX4 and Gazebo installed