PX4/PX4-Autopilot #5394: Snapdragon Flight DriverFramework driver and wrapper to support PWM based motor actuation

jywilson, 2016-08-27:
Hello PX4 developers,

I have just posted a beta release of a new Snapdragon Flight Addon that provides an aDSP firmware image ("static image") capable of generating PWM signals within 5 usecs of accuracy.  GPIO lines are used to generate the signals in software, and the signal generation occurs with an ISR context to assure deterministic response.  Limited testing has been conducted using a hacked up PX4 ESC driver, but our test results were encouraging.  Test flights on my 200mm quadcopter looked stable and the additional overhead on the aDSP did not appear to degrade PX4 performance.

What we need now is a full fledged DriverFramework driver and wrapper to consume the right uORB's and make the right DSPAL calls to generate the PWM signals for the ESC's.  The updates to the DSPAL interface are available here: https://github.com/ATLFlight/dspal.  The PWM interface is described in the include file: dev_fs_lib_pwm.h.

Are there any Snapdragon Flight users in the community available to assist in this effort?

Thanks,

Jim W.

twistxj, 2016-09-22:
Is there any update for this?

ohlarikd, 2016-10-01:
We are not quite ready to test, but this is something that we are very interested in having.  The Qualcomm ESC is obviously very restrictive to only small UAVs.  However, we will start looking at what you have!

HW-wise, are you sending the GPIO directly from Flight to the 'standard' 3-pin ESC's that out there?  So I just need to make a little adapter board?

sunshine-yan, 2016-12-01:
@jywilson 
Hi,I have updated the latest Snapdragon Flight Addon,but i have clicked the ‘calibration’ button  when power setup,calibration failed and prompt for errors:Can't open PWM devices.Please give me some suggestions,Thanks!
Error:
calibration started: 2 esc 0097 esc_calibration.cpp
calibration failed: Can't open PWM device 0177 esc_calibration.cpp

julianoes, 2016-12-02:
@Stifael is working on this, stay tuned.

venuannamraju, 2016-12-05:
@jywilson
Do you see any value in using a $20, 6-channel PWM converter with 0.25us resolution (Pololu maestro 6 channel). We recently interfaced it with ESC UART.

julianoes, 2016-12-05:
@venuannamraju I think there is nothing wrong with a converter but PWM directly out of Snappy is a clean integrated solution.

PR is now: #5940

venuannamraju, 2016-12-05:
Oh absolutely! Leveraging all that the board can provide makes best sense.
I brought it up as I did not see enough GPIOs unless we used the extension pins, and that SW PWM had 5usec jitter.

closed 2016-12-11 by LorenzMeier