PX4/PX4-Autopilot #5397: LPE: Allow althold-only mode

LorenzMeier, 2016-08-27:
@jgoppert It seems that a couple of users have issues entering altitude hold mode alone. Have you verified recently that the estimator is initialising without GPS for altitude?

Alirezam96, 2016-08-27:
http://logs.uaventure.com/view/55Ajk6mUiQByesrysrQsXf#_PLOT

LorenzMeier, 2016-08-27:
@jgoppert Test log: http://logs.uaventure.com/view/55Ajk6mUiQByesrysrQsXf#_PLOT

Alirezam96, 2016-08-27:
that log file is it ok? or i send another document ?

jgoppert, 2016-08-27:
@LorenzMeier The issue is that the LPE_GPS_ON flag = 1. In this case it will require a GPS lock before initializing altitude.

This change was made to fix the startup issue where there would be an altitude offset from GPS if baro initialized before GPS. The alternative would be to allow the estimator to shift the global altitude estimate once good GPS is detected for the first time.

I agree that it is confusing, if you want me to implement the altitude shift once GPS is detected let me know. If I just do it so that local z remains the same, but the global z changes, it might not be that huge of an impact.

jgoppert, 2016-08-27:
Another thought would be to only require the GPS lock for altitude if the gps has been detected, this would be the best approach I think, but how best to detect that rapidly. Would it be enough to look for a non-zero timestamp on the subscription, could we always count on that getting a message before baro tries to initialize?

LorenzMeier, 2016-08-27:
Neither approach is really useful. PX4 has the concept of a monotonic Z altitude, but its not implemented consistently yet. The baro altitude should be tracked monotonically, the global altitude should be tracked as offset to this altitude.

Right now we have logic in place to reset the Z altitude setpoint in the controller. So please have a look at INAV and implement the equivalent until we managed to rework the complete controller data path for altitude.

jgoppert, 2016-08-27:
INAV only publishes global position when it has a GPS estimate. Before that, it publishes the local position with reference altitude zero. After it gets GPS it changes the reference altitude to the GPS altitude. So, this is equivalent to me doing altOrigin reset on first gps initialization, won't be hard for me to implement.

LorenzMeier, 2016-08-27:
I think we should do it this way then. I'd like to change later on to publish only the altitude message in that state, but I don't want to take the control system apart right now.

Alirezam96, 2016-08-27:
thanks , i understand from discussion that if i lock GPS once , local z position set and i can use barometer without GPS ?

jgoppert, 2016-08-27:
@Alirezam96 currently you need to set LPE_GPS_ON = 0, then it will allow you to fly in altitude hold mode without GPS.

jgoppert, 2016-08-27:
See PR here: https://github.com/PX4/Firmware/pull/5398

closed 2016-09-01 by jgoppert