PX4/PX4-Autopilot #5399: Preflight checks: Improve user feedback

LorenzMeier, 2016-08-27:
We need to rework how we do pre-flight checks and user reports to be based on a reporting scheme that generates messages based on system states on the UI side.

LorenzMeier, 2016-08-27:
@DonLakeFlyer This will also need a QGC counterpart.

boosfelm, 2016-09-21:
as discussed, please take @arminambuehl in the loop

LorenzMeier, 2017-04-10:
I've improved a bunch, but still needs polishing.

bkueng, 2017-06-29:
Adding an example: When trying to arm, QGC just showed me 'REJECT POSITION CONTROL'. I had valid GPS, but ekf2 had no valid position. After some debugging we found that the range sensor was too close to the ground and thus ekf2 did not initialize.
The user feedback should be improved here.

closed 2018-02-21 by PX4BuildBot