PX4/PX4-Autopilot #5406: Snapdragon Flight launch failure (px4_log_initialize undefined symbol)

bharath374, 2016-08-29:
We get a run-time error and it appears to be due to this commit: https://github.com/PX4/Firmware/commit/eede43a786bbcfaf66ea91123247fdd50baa72e9

@bkueng Any idea? Did you test this on Snapdragon flight before/after your commit. We use the "make eagle_legacy_default" build target due to our configuration. Could you please confirm whether it was tested and works with the "make eagle_default" build target?

This is the mini-dm output we get on startup:

[08500/03]  05:50.527  HAP:63:undefined PLT symbol px4_log_initialize (209) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:50.591  HAP:63:undefined PLT symbol px4_log_initialize (209) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:50.623  HAP:63:undefined PLT symbol px4_log_initialize (209) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:50.655  HAP:63:undefined PLT symbol px4_log_initialize (209) /libpx4muorb_skel.so  0303  symbol.c

This is the apps proc output we get when I try to run px4:

creating new parameters file
creating new dataman file

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

ERROR [muorb] Initialize Error calling the uorb fastrpc initalize method..
ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
  MAV_BROADCAST: curr: 0ERROR [platforms__posix__px4_layer] krait update param 297 failed
 -> new: 1
INFO  [mavlink] mode: Normal, data rate: 2880 B/s on udp port 14556 remote port 14550
pxh> WARN  [mavlink] no broadcasting address found

bkueng, 2016-08-30:
No I did not test this on Snapdragon as I don't have the hardware, but I have an idea what's going wrong. Can you try to comment the line
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORBMain.cpp#L82
and see if that helps?

bharath374, 2016-08-31:
Yes that solves the problem! Could you please push a fix when you get a chance?

bkueng, 2016-09-01:
fixed on master

julien10, 2016-09-01:
When I comment that line, the error disappears but the px4 crashes instantly and I am forced to power cycle the board.

bharath374, 2016-09-02:
@bkueng We seem to have the same problem as @julien10.

bkueng, 2016-09-05:
Looks like a different issue then. @julianoes can you please look into this?

julianoes, 2016-09-28:
@bharath374 is this still an issue?

LorenzMeier, 2017-01-04:
Fixed.

closed 2017-01-04 by LorenzMeier