PX4/PX4-Autopilot #5407: Offboard mode (Position triplet setpoint) ignores specified coordinate frame

Pryre, 2016-08-30:
According to the Mavlink message, SET_POSITION_TARGET_LOCAL_NED, you should be able to specify coordinate_frame to either local NED or body NED setpoints (and offsets as well), to allow for things like position or velocity control in the body frame.

The issue stems from the Mavlink receiver, where the "coordinate_frame" variable is dropped. There should be a check in [mavlink_receiver.cpp around L809](https://github.com/PX4/Firmware/blob/15e50b26dc5f3cf773bb1824a265dc28bd383279/src/modules/mavlink/mavlink_receiver.cpp#L809) to see what frame the setpoint is specified in, then do the appropriate transformation, and then most likely send this as a local NED frame setpoint to the commander.

The [specifications detailed](http://mavlink.org/messages/common#SET_POSITION_TARGET_LOCAL_NED) are:
    MAV_FRAME_LOCAL_NED = 1
    (Should act as it currently does)

    MAV_FRAME_LOCAL_OFFSET_NED = 7
    (Should act as it currently does, but offset by the current position and heading of the mav in the local frame)

    MAV_FRAME_BODY_NED = 8
    (Should act as a location in the body frame, with NED aligned with true North. Not too sure on the usefulness of this)

    MAV_FRAME_BODY_OFFSET_NED = 9
    (Should tell the UAV to move with respect to it's current heading, i.e. a velocity command of 1m/s in the X direction should command the mav forward in the direction of it's current heading. A position setpoint should of [1,0,1] command the mav to move 1 meter forward, and 1 meter in altitude.)

A bit more information [can be found in the APM docs](http://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#copter-commands-in-guided-mode-set-position-target-local-ned) on what the expectations are. I am really only bothered by the lack of MAV_FRAME_BODY_OFFSET_NED, as it is quite desirable for moving a system with no regard for it's position in the world, but I'm sure others would like the other features as well.

I am willing to look into how to make the change myself, but I'm not sure how I should approach getting the data for the transformation into this section of the code.

dagar, 2017-04-11:
@pryre feel free to ping me if you'd like to implement this and want to talk about how it fits into the current architecture.

IeiuniumLux, 2017-04-26:
@pryre Thank you for pointing this out.  I was also wondering why the appropriate transformation was not happening when a coordinate_frame other than MAV_FRAME_LOCAL_NED was specified.

@dagar Will a PR be created to track and fix this?

mhkabir, 2017-04-26:
It would need to be implemented.

visrinivasan, 2017-11-15:
@mhkabir @dagar  Do you mean to say that setpoint target only supports FRAME_LOCAL_NED as of now and there's no support for FRAME_LOCAL_OFFSET_NED, FRAME_BODY_NED, FRAME_BODY_OFFSET_NED ?

mhkabir, 2017-11-15:
@visrinivasan Yes.

closed 2018-02-21 by PX4BuildBot

szebedy, 2018-04-21:
I would also need this feature for my vision based control algorithm. Can anyone suggest a workaround?

caomxin, 2019-03-22:
Need this feature!