PX4/PX4-Autopilot #5411: LPE: Position mode instability

mayanez, 2016-08-30:
Hi,

I was testing LPE today and had some problems with Position hold mode; the copter 'trembles' . The issue goes away when switching to ALT mode. 

Here are the links to the video:
https://drive.google.com/open?id=0B8AjOkamf22UYnRGUGl3cVhrWEU
and the log file:
https://drive.google.com/open?id=0B8AjOkamf22UT2M4OEtOc0M3UkE

Has anyone seen this behavior?

LorenzMeier, 2016-08-31:
Hi,

The issue is specific to your airframe and/or the RPI: http://logs.uaventure.com/view/mBXU7JvPygSgaqk47WjiSJ#IMU_Acc_PLOT

Your acceleration range is pretty significant. You have two choices (and you might need to apply both):

  * Check that you have enabled the on-sensor low pass filters for your accelerometer in driver framework. Make sure to have at least 40-80 Hz cutoff, if not lower.
  * Consider damping your autopilot on foam to reduce the incoming vibration

LPE (and EKF2) increase the accuracy of the estimate, but require a properly configured airframe. INAV in contrast never has great performance but works on a flying potato.

liaoytal, 2016-11-11:
Hi LorenzMeier:
 
   We also met the same problem. Optical Flow Indoor test.
   Position hold mode; the copter 'trembles' . The issue goes away when switching to ALT mode.

   The log file
https://drive.google.com/file/d/0B1tTxziAQDlnWFg4blNNTkV1RVU/view?usp=sharing

 We trying to modify some parameters 
 LPE_ACC_XY, LPE_FLW_XY,MPC_XY_P...
 But it is no use.

closed 2017-04-03 by mhkabir

kapoorshivam100, 2019-02-08:
In my case , in the position mode on increasing the throttle above 50 percent it stabilize  for a sec and then starts increasing altitude and throttle on his own...
what could be the posssible error??