PX4/PX4-Autopilot #5412: HOW to switch VTOL from MC to FW at release 1.4.1?

jarilq, 2016-08-30:
1.Just about to try the 1.4.1 and I cannot seem to find a way to program a transition switch.AUX1 does not work anymore like in 1.3.4,can you tell me how to switch from MC to FIXWING?

2.I make a vtol like QUADRANGE,and use airframe of quadrange and firmware 1.3.4 release.but when switch to TAKEOFF comand ,The UAV can takeoff and hold fix pos small second and then fly anywhere, flight controller can not control it,pitch and roll are big degree.Why?,I think the motor have some proble，do you think so？or maybe flight controller have some problem？

jarilq, 2016-08-31:
@JamesHarrison @LorenzMeier

sanderux, 2016-08-31:
it changed to RC_MAP_TRANS_SW
You need to set the value to the channel you want to use.

closed 2016-08-31 by sanderux