PX4/PX4-Autopilot #5423: HITL: Add aux control outputs

bartslinger, 2016-08-31:
I am setting up an interface between a Pixhawk and X-Plane using QGroundControl. I've noticed that HITL currently can only send 8 control commands over mavlink, which are the outputs of the main mixer.

My goal is to control a UAV with more than 8 actuators. Therefore I also need to use the 6 outputs of the AUX mixer. I would like to extend the HITL interface to also use values from the AUX mixer. From what I've seen so far, this is not trivial since pwm_out_sim only takes care of the main mixer through /dev/pwm_output0.

Of course this can be solved, but I would like to have some feedback before I implement it. When loading the aux mixer, the device can be set to /dev/pwm_output0 to redirect it to pwm_out_sim.cpp. This has to go in rc.interface:
```
if [ $OUTPUT_MODE == hil ]
	then
		echo "[HIL] Aux mixing redirected to pwm_output0"
		set OUTPUT_AUX_DEV /dev/pwm_output0
	else
		set OUTPUT_AUX_DEV /dev/pwm_output1
fi
```
However when this mixer is loaded, it resets the mixer in pwm_out_sim. I would like to remove [this reset](https://github.com/PX4/Firmware/blob/master/src/drivers/pwm_out_sim/pwm_out_sim.cpp#L746), so that the aux mixers are appended to the main mixers.

uORB message `actuator_outputs` already supports up to 16 actuators, so no changes are required for this. It does require some small changes in pwm_out_sim.cpp to actually fill these fields.

Then the final step (for the PX4-part) is to send this data out over mavlink. Therefore I need to add 6 channels in the HIL_CONTROL message.

Please let me know what you (hardcore developers?) think of this.

LorenzMeier, 2016-08-31:
Would this new message help you with this? If yes, what other fields would you like to see? https://github.com/mavlink/mavlink/pull/615

bartslinger, 2016-08-31:
That's perfect :+1: 
I had not seen that PR yet.

LorenzMeier, 2016-08-31:
ok, the PR is in. We will move our infrastructure to that now.

bkueng, 2016-09-01:
@bartslinger I integrated the mavlink message with 16 pwm's in https://github.com/PX4/Firmware/pull/5389 waiting for review. You can already look at it to see what's still missing for your use-case.

closed 2016-09-01 by LorenzMeier