PX4/PX4-Autopilot #5427: Reset yaw on arming

LorenzMeier, 2016-09-01:
Currently the yaw setpoint in manual control gets only reset during disarming. If you move the system in-between, it will return to its previous yaw.

zer1988, 2016-09-21:
Hi Lorenz,
I have one more faults relating to this one is that when my drone try to maintain its heading at the stationary point. I do not know why the value of IMU sensor increase or decrease its value 0.01/ sample time. Do you know how i can fix it? I re-calibrated my drone already but it still happens for manual mode.
If my drone's heading moves around the fixed point, this phenomenon stops.
Thanks. 
Phi Vu.

MaEtUgR, 2017-04-06:
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2366 looks to me like it resets on arming. I think we would have noticed that before if it would not.

mhkabir, 2017-04-06:
This is  _probably_ no longer an issue. Would be good if you could validate it.

closed 2017-04-10 by LorenzMeier