PX4/PX4-Autopilot #5429: I2C Sonar Problem

NCharabaruk, 2016-09-01:
Hello,

I am using a Pixhawk running the PX4 firmware. I am attempting to hook up a Maxbotix MB1220 sonar sensor for altitude determination. I've plugged the sensor into the I2C extender but nothing happens. There is no parameter in QGroundControl to enable the sensor. I know there is a driver for this MB12xx in PX4 and I've ruled out a hardware problem as I have used this sensor recently on an Arduino. Am I missing something?

Thanks

mhkabir, 2016-09-01:
You'll need to manually start the driver in the startup script (rcS) . We don't have it in by default.

NCharabaruk, 2016-09-01:
Thanks!

Is that stated somewhere? I've been trying to figure this out for a while and no one has been able to help.

closed 2016-09-01 by NCharabaruk

NCharabaruk, 2016-09-08:
For anyone else with this problem, drivers/mb12xx is commented out in the set(config_module_list section of cmake/configs/nuttx_px4fmu-v2_default.cmake. Uncommenting this line and then setting SENS_EN_MB12XX to 1 enables the sensor.

pakoman, 2016-12-07:
@NCharabaruk This is exactly what I've been looking for a while, thanks. 
Just an info, once I've done it where am I supposed to see the values in QGroundControl? Should I do some further steps in order to plot these distance values in the analyze tool?

Thanks

NCharabaruk, 2016-12-07:
To see the distance click on open the MAVLink Inspector widget and it is under Vehicle 1 -> Distance Sensor. You can plot the values in the analyse to if you want to. Only reason I could see to do that though is to verify that the values it is reading match the actual distance.

gsimon2, 2017-02-21:
I am having this same issue.

I have uncommented #drivers/mb12xx from  Firmware/cmake/configs/nuttx_px4fmu-v2_default.cmake and uploaded the firmware to the Pixhawk. However, I am unable to find SENS_EN_MB12xx in the full parameter list and am unable to get readings off of the sonar sensors.

Is there something that I am missing?

Thanks.