PX4/PX4-Autopilot #5433: vtol，Why not fly stable？

jarilq, 2016-09-01:
.I make a vtol like QUADRANGE,and use airframe of quadrange and firmware 1.3.4 release.but when switch to TAKEOFF comand ,The UAV can takeoff and hold fix pos small second and then fly anywhere, flight controller can not control it,pitch and roll are big degree.Why?,I think the motor have some proble，do you think so？or maybe flight controller have some problem？this is video http://v.youku.com/v_show/id_XMTcwOTYyMjQyOA==.html?from=y1.7-2，who can help me?

micoadmaster, 2016-09-02:
震动怎么样？PID是什么样的？看起来就是PID的问题可能性最大了

jarilq, 2016-09-02:
震动不大，偶尔飞的也挺好，用的官网的飞机和默认参数，难道还不能匹配，

jarilq, 2016-09-02:
# @ @micoadmaster

micoadmaster, 2016-09-02:
下日志然后分析一下就知道了，默认参数不会是这样的，肯定是有很大的误差才会导致这样

jarilq, 2016-09-02:
我从SD卡下的日志，为什么不能用地面站打开，你有qq或者微信吗？

jarilq, 2016-09-02:
@micoadmaster

mhkabir, 2016-09-02:
Please move this discussion over to discuss.px4.io since it seems to be  an airframe tuning issue, and not a core flightstack bug.

closed 2016-09-02 by mhkabir