PX4/PX4-Autopilot #5437: Camera PWM trigger issues

jcooper21, 2016-09-02:
@dagar @DonLakeFlyer I am not sure if this issue should be in PX4 firmware or QGC.  The main issue is we cannot set our PWM values when setting up camera trigger type in QGC.  All we can select is seagull QX1 and I do not have that camera to fly to test out survey.  We need to be able to setup our PWM values for different cameras manually as seen in mission planner.  We are missing our servo limits, Push and do not push values.

[image]

dagar, 2016-09-02:
@kellysteich @birchera Thoughts on adding this to camera_trigger?

AndreasAntener, 2016-09-02:
From a design perspective, the PWM camera interface implementation we have now should not be a MAP2 impl. but have a few parameters to specify PWM limits (neutral, on, off, shutter, AF, etc.).

Or, it needs to be abstracted more (for the architects under us ;)): the camera interface only considers how to output the control values (where it will have a PWM impl.). Then adding another interface that can be implemented to specify what values are used (where it will have a MAP2 impl.).

jcooper21, 2016-09-04:
@DonLakeFlyer @dagar I think on the QGC side this has to be supported either way as part of the APM compatibility otherwise no APM users will use QGC for surveying. Given it has to be built into QGC it would be great if PX4 firmware allowed for the same functionality.

DonLakeFlyer, 2016-09-04:
@jcooper21 Can you add a QGC issue with the same images for me to implement the ArduPIlot support?

jcooper21, 2016-09-11:
@dagar any chance this can fit into the PX4 1.5 release?

dagar, 2016-09-11:
I'm traveling this week, but if it's really as simple as pushed PWM value for duration, not pushed otherwise I can implement it when I get back. I'd have to lean on you for testing though as I don't have one of these devices. Interested?

jcooper21, 2016-09-11:
Ya that sounds good.

dagar, 2016-09-11:
Great!
Can you link to the device you use? What are reasonable defaults for pushed on, not pushed, and duration? What's "AC-T" mode?

jcooper21, 2016-09-11:
I use a few different devices linked below. The push, not pushed and duration values change for each device so we just need a blank field to input our numbers in. The AC-T is just a seagull feature so they set the ouch value to activate the the mode. 

http://www.seagulluav.com/seagull-map-x-support/#1445895077368-fc8aa683-2ecb

http://vp-systems.eu/order_cr.html

http://www.tuffwing.com/store/store.html

dagar, 2016-09-23:
@jcooper21 I'm just trying to figure out how to make this cleanly fit with what we already have (MAP2).

jcooper21, 2016-09-23:
What you currently have with map2 will not work.  Currently its setup incorrectly for most users.

dagar, 2016-09-23:
I think I know what you mean, but please elaborate. Which part is setup incorrect for most users?

Currently I'm creating another type of trigger (next to gpio and seagull) that's just simple PWM values on for duration, then off. Does that sound correct? I don't have or use any of these devices so you're going to have to be very specific if I've misinterpreted.

jcooper21, 2016-09-23:
That sounds correct another drop down for PWM and then be able to input values into the PWM fields as shown in the screenshot.  That will work perfectly.  Are you integrating the Camera feedback at the sametime?

dagar, 2016-09-27:
Are the hard coded values for Seagull correct or should it be configurable?

#define PWM_CAMERA_DISARMED			900
#define PWM_CAMERA_ON				1100
#define PWM_CAMERA_AUTOFOCUS_SHOOT	1300
#define PWM_CAMERA_NEUTRAL			1500
#define PWM_CAMERA_INSTANT_SHOOT	1700
#define PWM_CAMERA_OFF				1900
#define PWM_2_CAMERA_KEEP_ALIVE		1700
#define PWM_2_CAMERA_ON_OFF			1900

I'm just trying to determine if Seagull is a special case of PWM or can be within the same class.

jcooper21, 2016-09-27:
Those setting work for for seagull sony cameras but not for others for example camera on/off does not exist for other manufacturers even with seagull cables and instant shot can be changed to another mode if it was available and that is at a 1800 PWM value.

dagar, 2016-09-27:
So it sounds like we probably want a dropdown of predefined Seagull cables, but I'll leave that alone for now and just do a completely separate trigger type (within camera_trigger) for simple PWM on (duration) / off.

2 new params
TRIG_PWM_ON - PWM on value in us
TRIG_PWM_OFF - PWM off value in us
TRIG_ACT_TIME (already there).

mhkabir, 2017-04-03:
Closing in favour of #6961.

closed 2017-04-03 by mhkabir