PX4/PX4-Autopilot #5438: QGroundControl unbale to receive Mavlink messages

bharath374, 2016-09-02:
QGroundControl does not receive any Mavlink communication or telemetry data from Snapdragon flight anymore (other than a generic heartbeat). Communications from QGC to the Snapdragon Flight target are fine, only the other way does not seem to work. The issue seems to have been introduced here: https://github.com/PX4/Firmware/commit/15e50b26dc5f3cf773bb1824a265dc28bd383279

The issue seems to be that "orb_exists" always returns "true" until "orb_subscribe_multi" is called, but 
"orb_subscribe_multi" never has a chance to get called since "orb_exists" always returns "true".

@LorenzMeier

LorenzMeier, 2016-09-04:
@bharath374 Can you test this fix locally? https://github.com/PX4/Firmware/pull/5450 We should also sync on the drone type we want to use for Snapdragon so we can finalise testing.

bharath374, 2016-09-06:
@LorenzMeier Tried it out, works!

LorenzMeier, 2016-09-06:
Applied. Closing.

closed 2016-09-06 by LorenzMeier