PX4/PX4-Autopilot #5445: Need better test failure reports for CI

LorenzMeier, 2016-09-04:
@AndreasAntener We need to provide better readability of CI test failures. In particular instructions how to reproduce should be printed in the Jenkins console. A good start would be to at least have a minimal output (e.g. VTOL test failed, run ```make posix gazebo_standard_vtol```, ```commander takeoff``` and ```commander transition``` to verify basic operation).

Note that only a fraction of developers have ROS installed, so we either need alternate instructions like this one or move our testing to something more lightweight like Dronekit.

AndreasAntener, 2016-09-04:
The MAVROS test will always be ROS. Although one could ask the question if we really need to check via MAVROS and instead just do it with MAVLink. Strictly speaking we want to test the offboard interface, not MAVROS. However, last year I was very happy with that setup because it also spotted breakages coming from the MAVROS side..

The mission tests can potentially be executed normally and we can easily print instructions on how todo that on failure (can QGC still load the non-json mission format?).

LorenzMeier, 2016-12-26:
Done!

closed 2016-12-26 by LorenzMeier