PX4/PX4-Autopilot #5453: Ripples of changes to sensor pipeline

LorenzMeier, 2016-09-06:
@bkueng The changes to the sensor pipeline we did for 1.4 (mostly in the sensors app) still seem to have ripples. Here is the same airframe with 1.4 and below with 1.3:

*What is important to note is that the roll rate noise is higher with 1.4*. That's an unintended side effect. We might have more with acceleration.

http://logs.uaventure.com/view/QBjZQg5KPjBrXSQ44cNHh6#IMU_Acc_PLOT

Acceleration
[image]

Gyros
[image]

Rollrate
[image]

And here with 1.3:
http://logs.uaventure.com/view/ZCeJNa5zybYvodCtib4AP8

Acceleration
[image]

Gyros
[image]

Rollrate
[image]

dagar, 2016-09-06:
I've seen this too and haven't been quite sure what to think.

https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1108

tubeme, 2016-09-12:
There is something strange in 1.4. I'm still struggling to make a good PID tuning from scratch on the quadcopter and could not actually do it... Especially the YAW control is not present nor as RC input nor as stabilization response. It seems like the quad is on alcohol, laggy and not able to hold properly in > 2 m/s wind neither in Manual neither in Position or Altitude... We are still using 1.3 for our production tests.

LorenzMeier, 2016-09-12:
Can you share a log file?

tubeme, 2016-09-16:
@LorenzMeier Hi Lorenz, sorry for the delay with my reply. I just found that one of my motors was not right so exaggerating the problem. But I still have my point in what I said and I see a problem. I just think that it is not a good taste to explain a case with hardware malfunction of any sort. So first I will fix the problem, fly again and then I will come back here to report.

ecmnet, 2016-09-18:
Just to let you know, that I've also issues regarding position control (hold) which are not LPE related. Up to now, I stick with 1.4.1RC1.

dagar, 2016-09-18:
@LorenzMeier / @bkueng / @jgoppert - Is the core issue here setting accelerometer_m_s2 from integral / dt instead of the value from accel_report? Could you explain the choice?

    //using the value instead of the integral (the integral is the prefered choice)

https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1096

LorenzMeier, 2016-09-19:
The integral reflects *all* values from the last time delta (so has no aliasing until the sensor is aliasing itself internally) while the low-passed value can show a non-zero mean. This is depending on the frequency setup and sampling rates. The code you're reading is all correct.

What I'm looking for is not an error per se in the sampling pipeline but either a requirement of the estimator taking the data as input in terms of frequency spectrum or a change in the on-sensor analog filtering that worsened the results.

However, so far all setups that users reported had been horrible before as well.

ecmnet, 2016-09-19:
@LorenzMeier To clarify your last statement for us users, It would helpful, if you could publish some benchmarks or e.g. refer to a log regarding accelerometer noise / vibration..

LorenzMeier, 2016-09-19:
This is a log from a standard vehicle with normal vibration levels. If you have anything significantly higher your vehicle has a fundamental vibration issue.

http://logs.uaventure.com/view/x46eDbXxMAHzyWV6bPPBdj

ecmnet, 2016-09-19:
@LorenzMeier Thanks a lot!

zer1988, 2016-09-22:
Hi Lorenz,
I have one more faults relating to this one is that when my drone try to maintain its heading at the stationary point. I do not know why the value of IMU sensor increases or decreases its value 0.01/ sample time. Do you know how i can fix it? I re-calibrated my drone already but it still happens for manual mode.
If my drone's heading moves around the fixed point, this phenomenon stops.
Thanks. 
Phi Vu.

kd0aij, 2016-10-04:
@LorenzMeier What is the size and type of airframe in the log you posted as an example?
Also what kind of vibration isolation and which flight controller was used?

ecmnet, 2016-10-06:
Still I have some serious issues with latest master (LPE) and Altitude hold, although my vibration levels seem to be ok.

http://logs.uaventure.com/view/pWV5Kyw4Hnkxq4ETSC62zi

[image]

In LPE I increased the StdDev of ACC_Z, which helped a bit when setting it e.g. to the maximum. With 1.4.1RC1 I have again no issues.

jgoppert, 2016-10-07:
I will try flying lpe on master tomorrow and post the logs.

mhkabir, 2016-10-09:
@LorenzMeier @bkueng Are we any closer to fixing the excess vibrations issue? 
With the same vehicle and dampening setup, the excess accelerations post-1.3 causes my VIO setup to go berserk, which used to work fine pre-1.3.
The sensor data is overall very noisy and causes the bias estimation to diverge in my estimator. Would be great if we could look into this.

mhkabir, 2016-10-09:
For example, this is pre-1.3 :
[image]
This is on recent-ish master : 
[image]

LorenzMeier, 2016-10-09:
@mhkabir The excess vibration is on your frame - its real, it just has been hidden from you earlier. You would need to pre-filter it for VIO. Its possible that we need to add a filtered topic for these applications.

But the real fix is to get the IMU mounting to a setup where it sees less than a G vibration if possible.

mhkabir, 2016-10-09:
Interesting, thanks for the insight. I will look into better mounting. Possibly an isolated,  battery load dampened platform. 

What would be necessary to get a filtered topic working though, for short-term?

ecmnet, 2016-10-10:
I can confirm, that improving vibration isolation solves that issue.  I am able to achieve now alt-hold accuracy at +/-2cm (LPE with LIDAR) without any further tuning.

ecmnet, 2016-10-10:
@mhkabir Using the battery to increase inertia seems to be a good idea. Would you mind to share the setup?

mhkabir, 2016-10-13:
@ecmnet Yes, sure. I can try to show it with pictures once I'm back home.

I'm just switching to one of our other copters for the VIO testing - basically, the entire top plate is isolated from the motor plates using rubber dampers, and the top plate carries the battery mass, a Pixhawk 2 and the cameras. The dampers are tuned to suit the mass and dynamic response of the top plate.

mhkabir, 2016-10-13:
@LorenzMeier Any pointers for prefiltering the data?

melisab, 2016-11-13:
I tried again with the latest master a couple of days ago and I still cannot hover the standard VTOL plane in multicopter (ALT) mode. Any additional parameters that need to be setup perhaps?

mhkabir, 2016-11-13:
@melisab Logs?

julianoes, 2016-11-22:
Let's close this thread as soon as #5895 is resolved.

melisab, 2016-11-22:
Sorry for my late response. I had around 10 logs on that day so not sure which one I tested using the latest firmware. Is there any indicator in the log that can show the firmware version? 
Basically, I flew standard VTOL using early August firmware, then flashed the firmware using the latest master then switched to ALTCTL and the plane couldn't maintain its altitude properly.
Will it give different result if I use a fresh flight controller which means parameters from the previous version will not be maintained in the new firmware?

julianoes, 2016-11-22:
@melisab:
You can check the git hash of a log in FlightPlot in "LogInfo", and I believe http://logs.uaventure.com/ also shows it.

Params that you have changed are usually kept from one version to the next. If they are left at the default and the default (`param_define`) changes, they will get updated.

LorenzMeier, 2016-11-28:
I'm closing this in favour of #5895.

closed 2016-11-28 by LorenzMeier