PX4/PX4-Autopilot #5462: Raspberry Pi uORB Segfault

mayanez, 2016-09-06:
I'm seeing strange issues with the uORB templated interface on the Raspberry Pi. It is causing LPE to crash. I've tried debugging the issue but haven't been able to pinpoint the problem. Does anyone have any suggestions about what may be causing this?

Here is the backtrace for the behavior I'm seeing:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x764a1450 (LWP 2632)]
__strchrnul (s=s@entry=0xba16c "/%s/%s%d", c_in=c_in@entry=37)
    at strchrnul.c:39
39      strchrnul.c: No such file or directory.
(gdb) bt
#0  __strchrnul (s=s@entry=0xba16c "/%s/%s%d", c_in=c_in@entry=37)
    at strchrnul.c:39
#1  0x76ce9cec in __find_specmb (format=0xba16c "/%s/%s%d")
    at printf-parse.h:108
#2  _IO_vfprintf_internal (s=s@entry=0x7649f4a0, 
    format=format@entry=0xba16c "/%s/%s%d", ap=ap@entry=...) at vfprintf.c:1315
#3  0x76d1275c in _IO_vsnprintf (string=0x7649f5f8 "", maxlen=<optimized out>, 
    format=0xba16c "/%s/%s%d", format@entry=0x7649f77c "\230;\v", args=..., 
    args@entry=...) at vsnprintf.c:119
#4  0x76cf3308 in __snprintf (s=<optimized out>, maxlen=<optimized out>, 
    format=0xba16c "/%s/%s%d") at snprintf.c:33
#5  0x0005f2b0 in uORB::Utils::node_mkpath (
    buf=buf@entry=0x7649f5f0 "\214\017Jv\244\255\r", f=<optimized out>, 
    meta=meta@entry=0x14, instance=<optimized out>)
    at ../src/modules/uORB/uORBUtils.cpp:56
#6  0x0005f04c in uORB::DeviceMaster::ioctl (this=0xdad90, 
    filp=<optimized out>, cmd=<optimized out>, arg=1984558708)
    at ../src/modules/uORB/uORBDevices_posix.cpp:716
#7  0x0000fb4c in px4_ioctl (fd=42, cmd=2490368, arg=1984558708)
    at ../src/drivers/device/vdev_posix.cpp:239
#8  0x0005da3c in uORB::Manager::node_advertise (this=this@entry=0xdac10, 
    meta=meta@entry=0xd4bfc <__orb_optical_flow>, instance=instance@entry=0x0, 
    priority=priority@entry=0) at ../src/modules/uORB/uORBManager.cpp:329
#9  0x0005db28 in uORB::Manager::node_open (this=0xdac10, 
    f=f@entry=uORB::PUBSUB, meta=0xd4bfc <__orb_optical_flow>, 
    data=data@entry=0x0, advertiser=advertiser@entry=false, 
    instance=instance@entry=0x0, priority=priority@entry=75)
    at ../src/modules/uORB/uORBManager.cpp:399
#10 0x0005dcdc in uORB::Manager::orb_subscribe (this=<optimized out>, 
    meta=<optimized out>) at ../src/modules/uORB/uORBManager.cpp:232
#11 0x0005d1c4 in uORB::SubscriptionBase::SubscriptionBase (
    this=this@entry=0x7649f868, meta=<optimized out>, interval=10, 
    instance=<optimized out>) at ../src/modules/uORB/Subscription.cpp:85
#12 0x0005d21c in uORB::SubscriptionNode::SubscriptionNode (
    this=this@entry=0x7649f868, meta=<optimized out>, 
    interval=<optimized out>, instance=<optimized out>, list=0x7649f77c, 
    list@entry=0x7649f768) at ../src/modules/uORB/Subscription.hpp:140
#13 0x0005d5b0 in uORB::Subscription<optical_flow_s>::Subscription (
    this=0x7649f868, meta=<optimized out>, interval=<optimized out>, 
    instance=<optimized out>, list=0x7649f77c)
    at ../src/modules/uORB/Subscription.cpp:127
#14 0x0002a374 in BlockLocalPositionEstimator::BlockLocalPositionEstimator (
    this=0x7649f768)
    at ../src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp:180
#15 0x00028518 in local_position_estimator_thread_main (argc=<optimized out>, 
    argv=<optimized out>)
    at ../src/modules/local_position_estimator/local_position_estimator_main.cpp:155
#16 0x000a54e0 in entry_adapter (ptr=0xdd8a8)
    at ../src/platforms/posix/px4_layer/px4_posix_tasks.cpp:108
#17 0x76f8ce90 in start_thread (arg=0x764a1450) at pthread_create.c:311
#18 0x76d7b698 in ?? ()
```

LorenzMeier, 2016-09-06:
Are you sure you don't run in a stack smashing issue? Please try to increase the stack size by adding 5000 constantly here:
https://github.com/PX4/Firmware/commit/4a5eae23a29d452596ee0c8d0961f661ae635214

If that's the cause I recommend to add stack statically when building on Linux. I would start with 5K.

mayanez, 2016-09-06:
Yeah that's what was happening. Adding another 4K to PTHREAD_STACK_MIN seems to do the trick. I'm going to quickly flight test it and verify. 

I'll submit a PR for it soon.

julianoes, 2016-09-28:
@mayanez what's the state here?

mayanez, 2016-10-14:
Sorry for the late reply. If I'm not mistaken this has been resolved after Lorenz pushed a more generic fix.

julianoes, 2016-10-14:
Ok, thanks, closing then.

closed 2016-10-14 by julianoes