PX4/PX4-Autopilot #5465: SITL gazebo with IRIS broken

julianoes, 2016-09-07:
The simulation in gazebo seems broken on master.

Steps to reproduce:
```
make posix gazebo
```

Then try to takeoff using:

```
pxh> commander takeoff
```

julianoes, 2016-09-07:
Bisect points to 010c9e937b23988b222de6a1f714b57690f7742f

jgoppert, 2016-09-07:
This is odd, how did the SITL CI pass?

bkueng, 2016-09-07:
This works fine for me, I use gazebo 6.5.1.

julianoes, 2016-09-07:
It seems to be only my my Ubuntu 16.04 with gazebo7. Mac with gazebo7 is ok.

LorenzMeier, 2016-09-07:
Works fine on Mac here as well after all clean actions (make and git). I'm on stack_abstraction which is a branch addressing a few OS-specific stack size issues. It might be worthwhile for you to try as well.

julianoes, 2016-09-08:
Did a fresh checkout of Firmware and all is good again :confused:.

julianoes, 2016-09-08:
The file Tools/sitl_gazebo/models/iris/iris.sdf was not re-generated/updated.

The following resolved the problem:
```
cd Tools/sitl_gazebo
git clean -d -f -x
```

dagar, 2016-09-08:
@julianoes Try `make distclean` next time. I use it when jumping between branches with big changes, especially when submodules have been added/deleted/url changed.

julianoes, 2016-10-22:
This should be fixed by https://github.com/PX4/sitl_gazebo/pull/57, see https://github.com/PX4/sitl_gazebo/pull/57/commits/d9b9f5b96012bc90be2ceb05483788f6ae62dabd.

closed 2016-12-28 by LorenzMeier