PX4/PX4-Autopilot #5471: Parameters Missing after Firmware Update

BarzinM, 2016-09-08:
Hi,

I recently updated the PX4 firmware through QGC 3.0.0. I tried to set the values for some parameters found in https://pixhawk.org/firmware/parameters but received the following errors:

    [ERROR] [1449848580.090083069]: PR: Unknown parameter to get: INAV_W_Z_VIS_P
    [ERROR] [1449848604.929257906]: PR: Unknown parameter to get: INAV_W_XY_VIS_P
    [ERROR] [1449848629.545696700]: PR: Unknown parameter to get: INAV_W_MOC_P
    [ERROR] [1449848668.490732996]: PR: Unknown parameter to get: INAV_W_Z_BARO
    [ERROR] [1449849711.621053162]: PR: Unknown parameter to get: CBRK_NO_VISION

Trying some other random parameters:

    $ rosservice call /mavros/param/get ATT_EXT_HDG_M
    success: True
    value: 
      integer: 1
      real: 0.0
    $ rosservice call /mavros/param/get LPE_VIS_Z 
    success: True
    value: 
      integer: 0
      real: 0.5
    $ rosservice call /mavros/param/get LPE_VIS_XY
    success: True
    value: 
      integer: 0
      real: 0.5

Looking at parameters through QGC shows only `EKF2_REC_PRL` in `EKF2` section and only `INAV_LIDAR_ERR` in `Position Estimator INAV` section.

[image]

I tried this on a different Pixhawk and had the same results. Here is the firmware update output from QGC:

    QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
    All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
    Please unplug your Pixhawk and/or Radio from USB.
    Plug in your device via USB to start firmware upgrade.
    Plug in your device via USB to start firmware upgrade.
    Found device: Pixhawk
    Connected to bootloader:
    Version: 4
    Board ID: 9
    Flash size: 2080768
    Downloading firmware...
    From: http://px4-travis.s3.amazonaws.com/Firmware/stable/px4fmu-v2_default.px4
    Download complete
    MAV_AUTOPILOT = 12
    Succesfully decompressed parameter_xml
    Succesfully decompressed airframe_xml
    Unable to open airframe meta data file /home/naslab/.config/QGroundControl.org/PX4AirframeFactMetaData.xml for writing, error: Permission denied
    Succesfully decompressed image
    Erasing previous program...
    Erase complete
    Programming new version...
    Program complete
    Verifying program...
    Verify complete
    Upgrade complete
    ------------------------------------------

Any help is appreciated.

mhkabir, 2016-09-09:
Inav is no longer enabled by default, and hence it's parameters don't show up. 

LPE is the default position estimator for multicopter vehicles now.

BarzinM, 2016-09-09:
Great. Thanks @mhkabir for the reponse. I followed your hint and found this: http://dev.px4.io/advanced-switching_state_estimators.html

- Is LPE the preferred estimator to use with mocap?
- Is there a way to take the sonar, barometer, etc. out of its Kalman filter or should I just give them a large stddev?
- Is it correct to assume that I can send the position from any mocap system to `LPE_VIC_P` and it's not just for Vicon.

BarzinM, 2016-09-09:
I'm sending messages to `mavros/vision_pose/pose` and the values that I read from `/mavros/local_position/pose` are not close at all. I experimented maximizing the stddev for `LPE_LDR_Z`, `LPE_SNR_Z`, set the `LPE_VIS_XY` and `LPE_VIS_Z` to `0.01`, set `LPE_GPS_ON` to `0` and none has helped getting better position values.

The `ATT_EXT_HDG_M` has been `1`.

I'm not using `/mavros/mocap/pose` because of having the same issue as #5174.

BarzinM, 2016-09-11:
Forgot to mention in the previous comment that when I apply a large change to the values that I'm publishing to `mavros/vision_pose/pose`, It does in fact change the `/mavros/local_position/pose` values but they again do not match.

mhkabir, 2016-09-16:
Try using the mocap pose topic now, with latest master of Firmware. We've fixed a few things in LPE. 

Make sure external heading is set to mode 2 for mocap.

BarzinM, 2016-10-18:
The mocap doesn't timeout anymore but the values that I'm observing from `rostopic echo /mavros/local_position/pose` are noticeably different from the values that I'm sending to `/mavros/mocap/pose`. For example, right now the vehicle is stationary and I'm sending the following values through mocap topic:
    
    x: -0.16, y: 0.02, z: 0.15

Here are few consecutive values echoed from local_pose:

    ---
    header: 
      seq: 28745
      stamp: 
        secs: 1449845578
        nsecs: 753035264
      frame_id: fcu
    pose: 
      position: 
        x: 0.340512275696
        y: -1.20852172375
        z: -1.41807949543
      orientation: 
        x: 0.00188732496278
        y: 0.772159642628
        z: -0.605353185802
        w: 0.193166883137
    ---
    header: 
      seq: 28746
      stamp: 
        secs: 1449845578
        nsecs: 781851264
      frame_id: fcu
    pose: 
      position: 
        x: 0.256414264441
        y: -1.00386178493
        z: -1.14789938927
      orientation: 
        x: 0.00188457381176
        y: 0.772153448388
        z: -0.605350506761
        w: 0.193200063377
    ---
    header: 
      seq: 28747
      stamp: 
        secs: 1449845578
        nsecs: 821851264
      frame_id: fcu
    pose: 
      position: 
        x: 0.14307063818
        y: -0.727887928486
        z: -0.784500300884
      orientation: 
        x: 0.00187386724753
        y: 0.772145180571
        z: -0.605348321593
        w: 0.193240053527
    ---
    header: 
      seq: 28748
      stamp: 
        secs: 1449845578
        nsecs: 851851264
      frame_id: fcu
    pose: 
      position: 
        x: 0.0289213526994
        y: -0.448845684528
        z: -0.426012694836
      orientation: 
        x: 0.00186866179223
        y: 0.772134326015
        z: -0.605352265625
        w: 0.19327111837
    ---
    header: 
      seq: 28749
      stamp: 
        secs: 1449845578
        nsecs: 892669184
      frame_id: fcu
    pose: 
      position: 
        x: -0.00553367706016
        y: -0.364835113287
        z: -0.314119398594
      orientation: 
        x: 0.00186793178182
        y: 0.772134854759
        z: -0.605341171782
        w: 0.193303757447
    ---
    header: 
      seq: 28750
      stamp: 
        secs: 1449845578
        nsecs: 922669184
      frame_id: fcu
    pose: 
      position: 
        x: 0.0479081571102
        y: -0.494988560677
        z: -0.484442234039
      orientation: 
        x: 0.00185514201113
        y: 0.772137234325
        z: -0.605327606383
        w: 0.193336853104
    ---
 
It can be seen that the position values are not near the values that I'm sending and they are fluctuating too much.

BarzinM, 2016-10-18:
Maybe helps to mention that when I stop publishing to `mocap` topic, the `local_position` values diverge drastically.

LorenzMeier, 2017-01-04:
Closing this as stale. We are working (see pull requests) on updates to the vision interface pipeline.

closed 2017-01-04 by LorenzMeier