PX4/PX4-Autopilot #5483: Log files WRONG time stamp BUG

tubeme, 2016-09-12:
It is very old problem (>1 year old) that is not corrected. It is not critical bug but annoying.

When GPS lock is acquired then the time stamp of the Log files is GMT time, and not my zone which is +2 hours.  Sometimes it becomes 3 hours difference when the summer winter time changes.

mhkabir, 2016-09-12:
Not a bug really. It's a feature we need, to select the timezone's offset from GMT.

tubeme, 2016-09-12:
@mhkabir The bug is that it does not account for my timezone despite it has locked GPS. Of course having the GMT time is must and it works.

If there is a way we could manually set the time zone and summer/winter time offset it could be great. Or even better idea is to get the QGC host to send its timezone and s/w offset to PX4 when connected.

mhkabir, 2016-09-12:
Ah. Well, we should be able to implement this fairly easily. It's just not done. 

> Or even better idea is to get the QGC host to send its timezone and s/w offset to PX4 when connected.

That would be easy as well. See the SYSTEM_TIME mavlink message.

mhkabir, 2016-09-12:
> The bug is that it does not account for my timezone despite it has locked GPS.

It's not easy as I had thought. It would be almost impossible to fit a lookup table with timezones into the flash of an embedded STM32. Also, your GNSS receiver has no idea about your timezone, it just knows the UTC/GMT time.

tubeme, 2016-09-13:
The simple solution would be just to have a parameter with + or - hours, and summer winter time parameter. There is no need for implementing lookup table and time zones. Just for now. We have flights in Bulgaria EU and California and sometimes things get messy when we have to point out which log is flown when...

Next step could be just to sync the QGC host computer (tablet phone) timezone to PX4 parameters. It could be pretty easy to implement in QGC.

kd0aij, 2016-09-13:
see parameter SDLOG_UTC_OFFSET

tubeme, 2016-09-14:
Ups... my miss.

closed 2016-09-14 by tubeme

mhkabir, 2016-09-14:
Woohoo, my bad as well. @kd0aij Thanks for the tip!