PX4/PX4-Autopilot #5484: pre-commit hook differs from `make check_format`

julianoes, 2016-09-12:
I ran into the following case where the git pre-commit hook triggers but `make check_format` is all happy:

```
git commit
src/modules/mavlink/mavlink_mission.cpp:270: space before tab in indent.
+ 		} else {

make check_format 
Checking formatting with astyle
Format checks passed
```

FYI @AndreasAntener

AndreasAntener, 2016-09-12:
Awesome, isn't it ;)

The pre-commit hook includes the default git whitespace check which complained here. I guess we should included that in our check script instead.

julianoes, 2016-09-12:
Ah got, and yes it's cool! :smile:

dagar, 2016-09-17:
@julianoes does this catch it? #5503

dagar, 2016-09-20:
Merging #5503 because it's trivial. Let me know if it's not enough.

closed 2016-09-20 by dagar

julianoes, 2016-09-27:
Great, thanks @dagar!

PR #5503 (2016-09-20): make check_format also check git whitespace
-fixes #5484
commit c4b9b05a (2016-09-20): make check_format also check git whitespace (#5503)

-fixes #5484