PX4/PX4-Autopilot #5489: QuRT no longer built as part of Snapdragon flight target

bharath374, 2016-09-13:
The following make commands used to build both posix and qurt portions of the snapdragon flight build. Now they only build posix not wurt.
- make eagle_default
- make eagle_legacy_driver_default

I'm able to do the qurt build separately though:
- make qurt_eagle_default
- make qurt_eagle_legacy_driver_default

The issue seems to have been introduced during the recent makefile cleanup commits.
@CarloWood @bkueng

CarloWood, 2016-09-14:
Yes, thanks - I can reproduce this...

The problem seems to be that if you do:

foo% bar%:
	echo Hello $@

test: foo1 bar2

then 'foo% bar%' is considered to be a single target,
even when the stem is different, and hence only 'built'
once. When 'make test' is run, foo1 causes 'Hello foo1'
to be printed, but bar2 doesn't cause it the echo
to be executed again.

I can't really think of a way to fix that, short of 
splitting up the posix_% and qurt_% targets (and then
why not, the nuttx_% too), thus:

 # All nuttx, posix and qurt targets.
-nuttx_% posix_% qurt_%:
+nuttx_%:
+       $(call cmake-build,$@)
+
+posix_%:
+       $(call cmake-build,$@)
+
+qurt_%:
        $(call cmake-build,$@)

On Tue, 13 Sep 2016 15:57:41 -0700
BharathR <notifications@github.com> wrote:

> The following make commands used to build both posix and qurt
> portions of the snapdragon flight build. Now they only build posix
> not wurt.
> - make eagle_default
> - make eagle_legacy_driver_default [...]

-- 
Carlo Wood <carlo@alinoe.com>

CarloWood, 2016-09-14:
Hmm, now the weird thing is.. it SEEMS to do that (and the above patch should fix this), but when I actually test the following makefile:

foo% bar%:
        @echo "Complex build instructions for $@"

all: foo_a bar_b

and run: make -f themakefile all

it prints:

Complex build instructions for foo_a
Complex build instructions for bar_b

CarloWood, 2016-09-14:
Please continue discussion in #5491 (I couldn't find a way to add a pull request to this PR). This PR can thus be closed.

closed 2016-09-14 by LorenzMeier