PX4/PX4-Autopilot #5490: Adding A New Parameter

tanenyi, 2016-09-14:
I am trying to add a completely independent parameter into Pixhawk so I can use it in the running code for added functionality.

This is what I have done:

* Edit code in Firmware/src/modules/systemlib/system_params.c

```
/**
 * Enable Custom Function
 *
 * @value 0 disable
 * @value 1 enable
 * @min 0
 * @max 1
 * @reboot_required true
 * @group System
 */
PARAM_DEFINE_INT32(SYS_CSTM_FUNC_ENABLE, 0);
```

* Run make **px4fmu-v2_default** in CLI (Terminal)

Results:

* **parameters.xml** reflects the new parameter

* **nuttx-px4fmu-v2-default.px4** seems to be updated (time modified changes)

* After uploading the firmware to my Pixhawk and going into the Parameters List, the recently added parameter however does not show up.

Questions:

* Are there any more steps needed to make this work?

* Is there a convention to read/call up these parameters?

Firmware/src/drivers/px4fmu/fmu.cpp (L334), they are referenced directly:
```
for (unsigned i = 0; i < _max_actuators; i++) {
	_min_pwm[i] = PWM_DEFAULT_MIN;
	_max_pwm[i] = PWM_DEFAULT_MAX;
}
```

Firmware/src/drivers/camera_trigger/camera_trigger.cpp (L244), they are called with a function:
```
_p_interval = param_find("TRIG_INTERVAL");
_p_distance = param_find("TRIG_DISTANCE");
_p_activation_time = param_find("TRIG_ACT_TIME");
_p_mode = param_find("TRIG_MODE");
_p_interface = param_find("TRIG_INTERFACE");
```

* Where should I look if I want to override some of the sensor values with my own to feed into the Pixhawk? I have been looking at Firmware/src/drivers folder and although it looks to be the right place, it seems that there is a data entry point before that.

LorenzMeier, 2016-09-14:
For the parameter to show up in the GCS it needs to be used in the code (param find). It will then be in the default group.

LorenzMeier, 2016-09-14:
For sensors, you would need to copy one of the sensor drivers and publish the sensor data like they do.

closed 2016-09-14 by LorenzMeier