PX4/PX4-Autopilot #5497: posix_sitl hangs when gdb is attached

CarloWood, 2016-09-15:
Continuing a conversation that was started in #5443.

So far the following came out of the investigation:
* Things work when gdb is not used, and something locks up when gdb is used. This even happens when gdb is attached afterwards.
* Not only px4 becomes unresponsive (read: appears to be locked up), even gdb itself is often locked up (ie, doesn't react to control-C).
* Sending the px4 process SIGINT signals causes it to continue shortly; ie QGC reacts (sees the process) and I get a pxh> prompt, but immediately after things return to the locked-in state.

LorenzMeier, 2016-09-17:
@CarloWood Is this still an issue for you or did the updated docs here help?
http://dev.px4.io/simulation-debugging.html

CarloWood, 2016-09-21:
I'm still working on it! There is something wrong for sure, it's just very very hard to find.

CarloWood, 2016-11-06:
Well, this problem has gone away after upgrading from Ubuntu 14.04 to 16.04.  Not sure what caused that, could be version of gdb, or many other things...

CarloWood, 2016-11-06:
Just checked 4a5eae23a29d452596ee0c8d0961f661ae635214 and it now works, so definitely fixed as a result of my OS upgrade.

closed 2016-11-06 by CarloWood